pfalcon/esp-open-sdk #15: Please include the esp image tool - esptool-ck

kmpm, 2015-01-08:
https://github.com/tommie/esptool-ck

For use as described in 
https://github.com/esp8266/esp8266-wiki/wiki/Building instead of sdk/tools/gen_appbin.py

I know it can be installed using a deb but it would be nice to have a all in one solution to this sdk mess and you have been doing such a good work so far.

pfalcon, 2015-01-08:
esptool is already included - https://github.com/themadinventor/esptool , that's the one developed immediately after esp8266 went "public", the most known to the people, having the most features and easily hackable to add more features or resolve issues (because it's written to Python).  "Other" esptools are superfluous and lead to confusion.

closed 2015-01-08 by pfalcon

kmpm, 2015-01-08:
so themadinventor/esptool can create the files necessary for uploading as
well as upload them? I must have missed something in the documentation.

I thought they did 2 separate things,  prepare firmware and upload firmware
but had an unfortunate naming collision.

If I look at the makefile for blinkly example makefile,
https://github.com/esp8266/source-code-examples/blob/master/blinky/Makefile
they use both.

pfalcon, 2015-01-08:
> so themadinventor/esptool can create the files necessary for uploading as well as upload them? I must have missed something in the documentation.

Yes, I contributed such a change few months ago, and lately documentation was updated, too.

> I thought they did 2 separate things, prepare firmware and upload firmware but had an unfortunate naming collision.

Yeah, there was (is) lot of confusion around, the reason I decided to put together esp-open-sdk as a kind of executable documentation for myself and folks who may find that useful.

You can find example of minimal (and sufficient) makefile to build a project with tools included in esp-open-sdk in https://github.com/pfalcon/esp-open-sdk/issues/6 . I plan to add an example project to the repo, but so far haven't got to that.

kmpm, 2015-01-09:
Great!
Thanks...