pfalcon/esp-open-sdk #17: Fails to build on Cygwin

tz1, 2015-01-13:
At least two problems, first a dos2unix - git is adding carriage returns which causes problems (e.g. set in ./bootstrap fails).  Second, configure says 
configure: error: in `/home/.../esp-open-sdk/crosstool-NG':
configure: error: C compiler cannot create executables
Oh yes it can, but they end in ".exe" or something else so whatever test it is using is broken.

pfalcon, 2015-01-13:
> git is adding carriage returns

I doubt it's the case that *git* adds CRs, but feel free to report that to git project.

>  Second, configure says

"esp-open-sdk" project consists of a single Makefile, and doesn't have any "configure". So again, feel free to report the issue to an upstream project (crosstool-ng? gcc?).

kireevco, 2015-05-01:
It also fails to build on mingw. I'm trying to elaborate build process here https://github.com/kireevco/packages/tree/master/esp-open-sdk/windows

Currently mingw seems to have more future
```
10-env_mingw.cmd
20-build_mingw.cmd
```

Please feel free to contribute any code / thoughts.

kireevco, 2015-06-12:
@pfalcon Any hint on where to dig?
Thank you.

pfalcon, 2015-06-15:
@kireevco : Ok, if you ask:

1. I ran cygwin and friends on Windows ~8 years ago, and the best thing I can say to anyone else still doing that: don't, save your time, get Linux VM, if your really can't switch real hardware to it. So, make sure you have good reasons to do - like, you care about all poor folks in the world who are stuck on windows. For anyone individually, using a VM is easier.

2. Don't use mingw for anything serious. It's purpose is to provide to ANSI C API fro gcc (and Windows API of course). That's not enough for more or less serious software, POSIX API is needed. That's what cygwin provides, so use it.

3. If you have problem building crosstool-NG on cygwin, try latest upstream version from https://github.com/crosstool-ng/crosstool-ng . If it doesn't work, submit them a bug and work with them to resolve it.

4. If it works, submit a bug to https://github.com/jcmvbkbc/crosstool-NG asking to upgrade to latest version.

5. When the issue is fixed upstream, I'll just pull newer crosstool-NG version to esp-open-sdk, voila.

kireevco, 2015-06-15:
@pfalcon hey, thanks for responding :)
1. Same with me, I'm on a Mac currently, and it was surprisingly easier to get esp-open-sdk working (and hopefully people will be able too). Also, I like the idea of a docker container. Works good via boot2docker for me.
The only reason why I'm trying to build esp-open-sdk (and mainly xtensa toolchain) is for folks that use Windows (only) and for some reason won't switch to docker/vagrant/vm.
So hope you understand my intentions.

I've been spending all my free time recently on trying to build xtensa toolchain for windows. PAAAAIIN. Trying both mingw and cygwin [here](https://github.com/kireevco/packages/tree/master/esp-open-sdk/windows)...

The thing that kinda bothers me is that there is a prebuild toolchain [here](https://drive.google.com/folderview?id=0BzWyTGWIwcYQendHbWlsNUZpX0E&usp=drive_web#list) and [UDK](https://github.com/CHERTS/esp8266-devkit) and people keep using those instead of leaning toward open projects.

I will try the upstream crosstool-NG but from my latest experiments I was able to build crosstool-NG easily, and ct-ng xtensa step was failing. I'll get the log, and hopefully we could figure it out and drive people to esp-open-sdk so I could publish some [binary builds](https://bintray.com/kireevco/generic/esp-open-sdk/view) for Windows people too.

CHERTS, 2015-06-16:
Hi all
@kireevco in my [UDK](https://github.com/CHERTS/esp8266-devkit) I use gcc-xtensa and newlib-xtensa from @jcmvbkbc
see my [build script](https://github.com/CHERTS/esp8266-devkit/blob/master/Espressif/build-xtensa-lx106-elf.sh)

If you want to discuss an build esp-open-sdk for windows, write to me on [email](sleuthhound@gmail.com)

kireevco, 2015-10-22:
@CHERTS Thanks for the link. Yeah, I need to make esp-open-sdk compatible with windows.
@pfalcon Cygwin currently goes the furthest, and I was able to build crosstool-NG with some patches,
Right now I'm getting xgcc error at the final stage compiler part. Looks like it can't find some binaries, but they are there. I'll keep investigating:

```
[INFO ]  Installing final compiler
[ERROR]    xgcc: error: spawn: No such file or directory
[ERROR]    /home/User/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/include/valarray:1233:1: internal compiler error: Segmentation fault
[ERROR]    make[6]: *** [xtensa-lx106-elf/bits/stdc++.h.gch/O2g.gch] Error 1
[ERROR]    /home/User/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/unordered_set.h:1294:1: internal compiler error: Segmentation fault
[ERROR]    make[6]: *** [xtensa-lx106-elf/bits/stdc++.h.gch/O2ggnu++0x.gch] Error 1
[ERROR]    make[5]: *** [all-recursive] Error 1
[ERROR]    make[4]: *** [all] Error 2
[ERROR]    make[3]: *** [all-target-libstdc++-v3] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc/gcc.sh@595]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 46:09.80)
[46:12] / ct-ng:148: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/User/esp-open-sdk/crosstool-NG'
../Makefile:273: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/User/esp-open-sdk/crosstool-NG'
Makefile:270: recipe for target '/home/User/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/User/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```

kireevco, 2015-10-26:
@pfalcon any idea?

pfalcon, 2015-10-26:
> Looks like it can't find some binaries, but they are there.

Then it looks for them in wrong place, or just doesn't know where to find them (PATH). There's also "internal compiler error", which doesn't give cheerful anticipations.

What I'd do (besides stopping wasting time with cygwin) is try to build just a native gcc from source. If it fails, your cygwin setup is broken. Otherwise, I'd try to build some other crosscompiler directly with crosstool-NG, if it fails, crosstool-NG doesn't play well with cygwin, time to contact crosstool-NG upstream.

kireevco, 2015-10-26:
> besides stopping wasting time with cygwin

I hear you, I hear you... But what would be my other option here? Windows users are still there and clearly will benefit of esp-open-sdk... MinGW is even worse...

> What I'd do (besides stopping wasting time with cygwin) is try to build just a native gcc from source. If it fails, your cygwin setup is broken. Otherwise, I'd try to build some other crosscompiler [...]

Good point. Let me try to isolate the issue. Thank you. (Ping me if you could think about anything else - highly appreciated).

pfalcon, 2015-10-26:
> But what would be my other option here? Windows users are still there 

I usually say that setting up a Linux VM under Windows will be less pain and more benefit.

kireevco, 2015-10-26:
I agree, @pfalcon, yeah, when you control those - sure. But not the "Users"... You know...:)

bhundven, 2015-11-09:
> crosstool-NG doesn't play well with cygwin, time to contact crosstool-NG upstream
This is a work-in-progress. See: https://github.com/crosstool-ng/crosstool-ng/pull/239

kireevco, 2015-11-10:
> This is a work-in-progress. See: crosstool-ng/crosstool-ng#239

I was able to build crosstool-Ng on Cygwin, but final stage compiler build fails.

mingwandroid, 2015-11-14:
This should all be fixed in upstream crosstool-ng, but you must follow the documentation at:

https://github.com/crosstool-ng/crosstool-ng/blob/master/docs/C%20-%20Misc.%20tutorials.txt#L317-L361

JetstreamRoySprowl, 2017-07-20:
I tried building with Cygwin using the June 7 2017 esp-open-sdk and run into this, which I don't see any way around since setting Windows' NTFS to all case-sensitive sounds like a disaster:

cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/home/RoyS/workspace/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[ERROR]  Your file system in '/home/RoyS/workspace/esp-open-sdk/crosstool-NG/.build' is *not* case-sensitive!
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
[ERROR]  >>        called from: CT_TestAndAbort[scripts/functions@351]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@93]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 25009558:50.85)
[00:00] / make[2]: *** [ct-ng:152: build] Error 1
make[2]: Leaving directory '/home/RoyS/workspace/esp-open-sdk/crosstool-NG'
make[1]: *** [../Makefile:136: _toolchain] Error 2
make[1]: Leaving directory '/home/RoyS/workspace/esp-open-sdk/crosstool-NG'
make: *** [Makefile:129: /home/RoyS/workspace/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

jcbottorff, 2017-07-31:
Just a suggestion for folks trying to get the toolset working on Windows. If you run a recent version of Windows 10, you can install the Linux subsystem for Windows, which allows running the Ubuntu user-mode environment. You then just follow the Linux instructions, and generate Linux binaries. This is NOT cygwin or other port of the tools, it's running the EXACT Linux binaries. Microsoft is shipping Windows kernel code that emulates most of the Linux kernel calls, allowing a process to think it's running on Linux. The only less clear part is you have to install a number of dependent packages, and I didn't offhand see a complete list. I've successfully built the toolset. See https://msdn.microsoft.com/en-us/commandline/wsl/install_guide for installing Ubuntu on Windows.