pfalcon/esp-open-sdk #19: Unable to use multiple jobs when compiling

pietrushnic, 2015-01-25:
```
make: *** No rule to make target '/home/pietrushnic/tmp1/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libc.a', needed by '/home/pietrushnic/tmp1/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libcirom.a'.  Stop.
make: *** Waiting for unfinished jobs....
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
Submodule 'crosstool-NG' (https://github.com/jcmvbkbc/crosstool-NG) registered for path 'crosstool-NG'
Submodule 'esptool' (https://github.com/themadinventor/esptool) registered for path 'esptool'
Submodule 'lx106-hal' (https://github.com/tommie/lx106-hal) registered for path 'lx106-hal'
Cloning into 'crosstool-NG'...
remote: Counting objects: 25635, done.
remote: Compressing objects: 100% (10653/10653), done.
remote: Total 25635 (delta 14470), reused 25635 (delta 14470)
Receiving objects: 100% (25635/25635), 14.88 MiB | 302.00 KiB/s, done.
Resolving deltas: 100% (14470/14470), done.
Checking connectivity... done.
Submodule path 'crosstool-NG': checked out '7c6bc14de33e7a331ad4932bb505385ba0963eea'
Cloning into 'esptool'...
remote: Counting objects: 152, done.
remote: Total 152 (delta 0), reused 0 (delta 0)
Receiving objects: 100% (152/152), 46.86 KiB | 0 bytes/s, done.
Resolving deltas: 100% (72/72), done.
Checking connectivity... done.
Submodule path 'esptool': checked out 'e1abbf64abe76513ad6e66bb34b7b55229198355'
Cloning into 'lx106-hal'...
remote: Counting objects: 46, done.
remote: Total 46 (delta 0), reused 0 (delta 0)
Unpacking objects: 100% (46/46), done.
Checking connectivity... done.
Submodule path 'lx106-hal': checked out '90cdf910a8431675b6814ba447550edca3a78d4c'
```

Any idea what should be fixed to use multiple CPU jobs when compiling ? Why it complains about libc.a and libcirom.a ?

sej7278, 2015-01-26:
its only the actual compiling you'd want multiple jobs for, git cloning and bootstrapping you want done one step at a time otherwise you may start compiling before you've downloaded all the files, such as the .a file you mentioned

pfalcon, 2015-01-28:
Yes, as @sej7278 says, it doesn't make much sense to parallelize esp-open-sdk's Makefile itself, you probably want to parallelize gcc, etc. build. That's handled by "ct-ng" tool, and looking at its --help, it accepts usual make's -j.

I personally don't fancy -j stuff, as prefer my laptop stay responsive and not whirl with a fan, but if you want to allow usage like `make CT_NG_JOBS=4`, while keeping default at 1, feel free to.

pietrushnic, 2015-01-28:
Sure, I mean parallel compilation for `crosstool-NG` usually I used `./ct-ng build.16` which use 16 jobs. `ct-ng` build is time consuming it takes about 20min on my i7 laptop with one job so using multiple jobs would be great, I will try `make CT_NG_JOBS=16` and will see what the result would be.

pfalcon, 2015-01-28:
Just in case if it wasn't clear from the previous message - before trying, you need to implement handling on that make var.

closed 2015-02-18 by pfalcon