pfalcon/esp-open-sdk #22: Branch for sdk-0.9.4

eadf, 2015-02-04:
I have some strange issues with sdk 0.9.5 on my system. 
Whenever i try to use a timer  (os_timer_disarm, os_timer_setfn, os_timer_arm) it never fires. 
Same code works fine in 0.9.4.

Until i figure out what's wrong, could you create a branch sdk-0.9.4?

eadf, 2015-02-04:
I just tried to run the blinky example on sdk-0.9.5 (https://github.com/esp8266/source-code-examples)
And it fails in the same way as my own code, nothing happends in the timers.
I'm using a command line environment on an up-to-date linux-mint 17.1. (sdk-0.9.4 works fine)

pfalcon, 2015-02-04:
There's no need for a branch, you can build a version with older SDK with:

~~~~
make VENDOR_SDK=0.9.4
~~~~

closed 2015-02-04 by pfalcon