pfalcon/esp-open-sdk #23: cant build on ubuntu14.04 x64

alonewolfx2, 2015-02-10:
i have this error message. 

```
xtensa-lx106-elf-gcc -DHAVE_CONFIG_H -I.   -I../include  -mlongcalls -MT debug_hndlr.o -MD -MP -MF .deps/debug_hndlr.Tpo -c -o debug_hndlr.o debug_hndlr.S
debug_hndlr.S: Assembler messages:
debug_hndlr.S:89: Error: unknown opcode or format name 'wsr.ICOUNT'
debug_hndlr.S:99: Error: unknown opcode or format name 'wsr.IBREAKENABLE'
make[4]: *** [debug_hndlr.o] Error 1
make[4]: Leaving directory `/home/mb/esp-open-sdk/lx106-hal/src'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/home/mb/esp-open-sdk/lx106-hal/src'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/home/mb/esp-open-sdk/lx106-hal'
make[1]: *** [_libhal] Error 2
make[1]: Leaving directory `/home/mb/esp-open-sdk/lx106-hal'
make: *** [/home/mb/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libhal.a] Error 2
```

supergis, 2015-02-11:
I build on ubuntu 14.04 X64 OK.
Please check your gcc version may get some usefull information.
Ensure run:
```
sudo apt-get install make unrar autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python sed
```
to install needed library.

pfalcon, 2015-02-11:
ubuntu 14.04 x64 is also my main setup where I built project from the beginning. But maybe I need to rebuild it from scratch some time - maybe there were recent upstream changes (can't do that right away as I'm travelling).

alonewolfx2, 2015-02-11:
I already installed needed library.i tried 3 times but i cant build.

SpaceTeddy, 2015-02-13:
I use Ubuntu 14.04.1 x64 as well. I set up a clean install in a Fusion VM. I simply clone the git rep and it worked almost out of the box. The only thing I have made ist to export the path 
"export PATH=/home/ubuntu/esp-open-sdk/xtensa-lx106-elf/bin:$PATH"
manually again.
after cloning nodemcu and make it compiles w/o any problems.

pfalcon, 2015-02-22:
I just rebuilt the project from scratch on my Ubuntu 14.04 x64 without any issues.

One hunch I may have about your issue is that you have some older xtensa toolchain in your PATH and it intervenes. Just in case, don't do that, enable each toolchain only for the session (i.e. console) it is needed for. And otherwise, way to investigate the issue would be brining up clean 14.04 install (e.g. in VM) and try there.

alonewolfx2, 2015-02-22:
i tried new ubuntu install on virtualbox and try again but same issue. i have no idea.

ystrem, 2015-02-22:
I build it in Linux Mint 17 x64 (whitch is Ubuntu 14.04 x64), and all good. (Fresh install in vmware player)

alonewolfx2, 2015-02-22:
can you upload builded folder somewhere

ystrem, 2015-02-22:
Sure, It's mean to me right ? :D

alonewolfx2, 2015-02-22:
yes. @ystrem if you can i will be happy.i didnt compile it.

ystrem, 2015-02-22:
@alonewolfx2 Ok I'm uploading to mega, but it has 1.1GB so you have to wait 20min's. I have only 10Mb UL :D

alonewolfx2, 2015-02-22:
10mb ul :) i am waiting. my dl is 16mb :))

ystrem, 2015-02-22:
@alonewolfx2 Ok, here it's https://mega.co.nz/#!uYc02ZRZ!XOurcBEDPIvKudehjCjlAzzHT4rMuff6WUOJ3JJTkQs :)

supergis, 2015-02-25:
@alonewolfx2  
Maybe the Ubuntu need update, can try run: 
```
sudo apt-get update 
sudo apt-get upgrade
```
to upgrade to latest library.

alonewolfx2, 2015-03-03:
I tried on fresh and updated Ubuntu x64 but result is same. I am using Turkish language  on Ubuntu. Is it matter for sdk? I can't figure that out this problem

pfalcon, 2015-03-12:
@alonewolfx2: Theoretically, that can be an issue, so I suggest setting English for that session and trying again. That should be as simple as running "export LANG=en_US.UTF-8" before any other step, but do you own googling.

liquiddandruff, 2015-03-20:
Clean install with successful make with no issues on xubuntu 14.04.

simos, 2015-04-08:
Compiled successfully on Ubuntu 14.04.2 LTS, x86_64.

con-f-use, 2015-09-25:
I get an "esp-open-sdk/crosstool-NG/.build/src/binutils-2.24/gas/as.c:224:14: error: 'TARGET_ALIAS' undeclared (first use in this function)" error.

>>  Build failed in step 'Installing binutils for host'
>>  Error happened in: CT_DoExecLog[scripts/functions@257]

Quentinb, 2016-04-27:
Hi All, I am trying this setup on a clean Ubuntu 16.04 TLS and this package does not build. I found that the binutils-2.24, gcc-4.8.2 and gdb-7.5.1 files do not exist on the expected website paths and are therefor not downloaded correctly resulting in error messages! I have manually downloaded and extracted them and now it fails with installing ISL. I am now out of ideas as I have spent a full day trying to get it to work.

I have followed the steps, one by one! If anybody has any ideas please help!!

jrobeson, 2016-10-20:
@Quentinb : that's not related to this issue. If you're still having this problem, then submit it as its own issue.

polyGithub, 2018-02-27:
poly@poly:~/esp-open-sdk$ make
Makefile:184: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:179: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory `/home/poly/esp-open-sdk/crosstool-NG'
../Makefile:184: warning: overriding commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:179: warning: ignoring old commands for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... no
checking for cc... no
checking for cl.exe... no
configure: error: in `/home/poly/esp-open-sdk/crosstool-NG':
configure: error: no acceptable C compiler found in $PATH
See `config.log' for more details
make[1]: *** [_ct-ng] Error 1
make[1]: Leaving directory `/home/poly/esp-open-sdk/crosstool-NG'
make: *** [crosstool-NG/ct-ng] Error 2