pfalcon/esp-open-sdk #24: wget failure

xxxajk, 2015-02-14:
cp esptool/esptool.py /home/oldfart/esp-open-sdk/xtensa-lx106-elf/bin/
wget --content-disposition "http://bbs.espressif.com/download/file.php?id=189"
--2015-02-13 21:24:41--  http://bbs.espressif.com/download/file.php?id=189
Resolving bbs.espressif.com... 192.185.229.242
Connecting to bbs.espressif.com|192.185.229.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7190185 (6.9M) [application/octet-stream]
--2015-02-13 21:24:42--  http://bbs.espressif.com/download/file.php?id=189
Reusing existing connection to bbs.espressif.com:80.
HTTP request sent, awaiting response... 200 OK
Length: 7190185 (6.9M) [application/octet-stream]
Saving to: `file.php?id=189.1'

100%[======================================>] 7,190,185   1.98M/s   in 4.4s

2015-02-13 21:24:46 (1.56 MB/s) - `file.php?id=189.1' saved [7190185/7190185]

unzip -q -o esp_iot_sdk_v0.9.5_15_01_23.zip
unzip:  cannot find or open esp_iot_sdk_v0.9.5_15_01_23.zip, esp_iot_sdk_v0.9.5_15_01_23.zip.zip or esp_iot_sdk_v0.9.5_15_01_23.zip.ZIP.
make: *** [esp_iot_sdk_v0.9.5/.dir] Error 9

wget needs the -o option to fix this.

xxxajk, 2015-02-14:
After manual renaming:
p esptool/esptool.py /home/oldfart/esp-open-sdk/xtensa-lx106-elf/bin/
unzip -q -o esp_iot_sdk_v0.9.5_15_01_23.zip
mv License esp_iot_sdk_v0.9.5
touch esp_iot_sdk_v0.9.5/.dir
ln -snf esp_iot_sdk_v0.9.5 sdk
wget --content-disposition "http://bbs.espressif.com/download/file.php?id=190"
--2015-02-13 21:28:24--  http://bbs.espressif.com/download/file.php?id=190
Resolving bbs.espressif.com... 192.185.229.242
Connecting to bbs.espressif.com|192.185.229.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 33774 (33K) [application/octet-stream]
--2015-02-13 21:28:25--  http://bbs.espressif.com/download/file.php?id=190
Reusing existing connection to bbs.espressif.com:80.
HTTP request sent, awaiting response... 200 OK
Length: 33774 (33K) [application/octet-stream]
Saving to: `file.php?id=190'

100%[======================================>] 33,774       179K/s   in 0.2s

2015-02-13 21:28:25 (179 KB/s) - `file.php?id=190' saved [33774/33774]

unzip -q -o sdk095_patch1.zip
unzip:  cannot find or open sdk095_patch1.zip, sdk095_patch1.zip.zip or sdk095_patch1.zip.ZIP.
make: *** [.sdk_patch_0.9.5] Error 9

pfalcon, 2015-02-18:
--content-disposition option to wget exactly takes care of correct downloaded file naming. It works as expected for example on Ubuntu 14.04 with wget 1.15. So, you likely have problems with your wget, and esp-open-sdk can't help in any way with that, sorry. Please bring that up with providers of your wget (like your OS vendor).

So, closing, but feel free to mention your OS/wget version, so other people who may have such issue had a chance to find it.

closed 2015-02-18 by pfalcon

xxxajk, 2015-02-20:
GNU Wget 1.11.4
I'll grab a newer source and compile. You may want to make a note someplace of the lowest version of wget supported.