pfalcon/esp-open-sdk #30: Doesn't build on Slackware-current

bogdanr, 2015-02-27:
When trying to build on Slackware Current I get this error message:

```
[DEBUG]    ==> Executing: 'touch' '/home/bogdan/esp-open-sdk/crosstool-NG/.build/src/.binutils-2.24.extracted'                                                                                                                                                                    
[DEBUG]    ==> Executing: 'rm' '-f' '/home/bogdan/esp-open-sdk/crosstool-NG/.build/src/.binutils-2.24.extracting'                                                                                                                                                                 
[DEBUG]    Entering '/home/bogdan/esp-open-sdk/crosstool-NG/.build/src/binutils-2.24'                                                                                                                                                                                             
[EXTRA]    Patching 'binutils-2.24'
[DEBUG]    Looking for patches in '/home/bogdan/esp-open-sdk/crosstool-NG/lib/ct-ng.1.20.0/patches/binutils/2.24'...
[DEBUG]    ==> Executing: 'patch' '--no-backup-if-mismatch' '-g0' '-F1' '-p1' '-f' '-i' '/home/bogdan/esp-open-sdk/crosstool-NG/lib/ct-ng.1.20.0/patches/binutils/2.24/120-sh-conf.patch'
[ALL  ]    patching file configure
[ALL  ]    Hunk #1 succeeded at 3793 (offset 223 lines).
[ALL  ]    /usr/bin/patch: listing attributes of configure: Invalid argument
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Patch[scripts/functions@1098]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@32]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@607]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 8:54.56)
```

pfalcon, 2015-02-27:
This should be reported to crosstool-ng and slackware-current.

klukonin, 2015-04-14:
Confirmed.

The same issue on ubuntu 14.04 x64

jrobeson, 2016-10-19:
Then shouldn't this ticket be closed as per https://github.com/pfalcon/esp-open-sdk/issues/101?

pfalcon, 2016-10-19:
I leave tickets I can't help with open, for other users to chime in and possibly help or at least confirm an issue. There was no activity in this ticket for a while, let's close it.

closed 2016-10-19 by pfalcon