pfalcon/esp-open-sdk #31: Problems with malloc,free and realloc (or os_xxxx counterparts)

MicioMax, 2015-03-01:
Using one of them (either including sdk mem.h or xtensa malloc.h) I get following error : 

undefined reference to `_sbrk_r'

Is there a way to have dynamic allocation with the sdk ?

pfalcon, 2015-03-01:
> Is there a way to have dynamic allocation with the sdk ?

Yes, by using SDK functions - refer to vendor docs, headers, etc.

Regarding standard malloc() and friends, I don't know what to say - there's no way to implement them portably for all embedded targets/projects. So, their implementation is usually left for a specific project, per its own requirements, which may differ from other projects.

MicioMax, 2015-03-01:
Well, I looked into ESP8266 sdk (0.95 version) libs wth 'ar -t' and I didn't find any allocation function... weird.

pfalcon, 2015-03-01:
I don't know why you chose the hard way. Mere grep for "alloc" in SDK include dir will get you going. Anyway, these questions are a bit out of scope here, this project provides GCC-based toolchain for vendor SDK, but how to use vendor SDK is better answered somewhere else, e.g. on esp8266.com .

closed 2015-03-01 by pfalcon