pfalcon/esp-open-sdk #33: Espruino with Xtensa on ESP8266

aplikatika, 2015-03-12:
Hi guys!

I am in a mission to run Javascript on esp8266.

I struggled a lot and I managed to compile the Espruino with Xtensa, now all goes well, but I stuck with this linker segmentation overflow "section .text' will not fit in region iram1_0_seg'"

0x00000.bin 195kb
0x40000.bin 144kb

Is there any guru who would like to help me solving this issue?

Espruino: https://github.com/espruino

pfalcon, 2015-03-12:
Guru from crystal ball says: leave javascript mediocrity behind, join forces of light with MicroPython  https://github.com/micropython/micropython .

Otherwise, offtopic here.

closed 2015-03-12 by pfalcon

slaff, 2015-03-13:
@aplikatika Can you share your code? I had an idea to use Espruino as a base for PHP interpreter for esp8266.

aplikatika, 2015-03-13:
@slaff of course! here: https://github.com/aplikatika/Espruino-on-ESP8266