pfalcon/esp-open-sdk #41: "Malformed archive" error building on Mac OS X 10.10

jgarbers, 2015-04-03:
I'm trying to install the SDK on my Mac, but have gotten this error twice now:

    Making install in src
     /usr/local/bin/gmkdir -p '/Volumes/Case/lab/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib'
     /usr/local/bin/ginstall -c -m 644  libhal.a '/Volumes/Case/lab/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib'
     ( cd '/Volumes/Case/lab/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib' && xtensa-lx106-elf-ranlib libhal.a )
    xtensa-lx106-elf-ranlib: libhal.a: Malformed archive
    make[4]: *** [install-libLIBRARIES] Error 1
    make[3]: *** [install-am] Error 2
    make[2]: *** [install-recursive] Error 1
    make[1]: *** [_libhal] Error 2
    make: *** [/Volumes/Case/lab/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libhal.a]     Error 2

I am installing on a case-sensitive drive image, but I used a sparsebundle instead of the fixed 10G image described in the documentation.  Could that cause my problem, or is there something else I can check?

jgarbers, 2015-04-03:
Looks like this is the same problem mentioned here: https://github.com/tommie/lx106-hal/pull/6

MacDada, 2015-04-13:
I have exactly the same error :/

pfalcon, 2015-04-13:
There were some fixes in lx106-hal for that, I now pulled those updates to esp-open-sdk, please make a new clean git clone and rebuild and see if that helped.

MacDada, 2015-04-13:
@pfalcon It's OK now, thx!

pfalcon, 2015-04-13:
Thanks for confirmation, closing then.

closed 2015-04-13 by pfalcon