pfalcon/esp-open-sdk #43: Unexpected Build Failure on MACOSX

steowens, 2015-04-11:
```
se-steowens-mb1:case-sensitive sowens$ git clone https://github.com/pfalcon/esp-open-sdk.git
Cloning into 'esp-open-sdk'...
remote: Counting objects: 147, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 147 (delta 0), reused 0 (delta 0), pack-reused 144
Receiving objects: 100% (147/147), 30.10 KiB | 0 bytes/s, done.
Resolving deltas: 100% (81/81), done.
Checking connectivity... done
se-steowens-mb1:case-sensitive sowens$ cd esp-open-sdk/
se-steowens-mb1:esp-open-sdk sowens$ ls
Makefile			c_types-c99.patch		crosstool-config-overrides	lx106-hal
README.md			crosstool-NG			esptool
se-steowens-mb1:esp-open-sdk sowens$ make STANDALONE=n
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
Submodule 'crosstool-NG' (https://github.com/jcmvbkbc/crosstool-NG) registered for path 'crosstool-NG'
Submodule 'esptool' (https://github.com/themadinventor/esptool) registered for path 'esptool'
Submodule 'lx106-hal' (https://github.com/tommie/lx106-hal) registered for path 'lx106-hal'
Cloning into 'crosstool-NG'...
remote: Counting objects: 25674, done.
remote: Total 25674 (delta 0), reused 0 (delta 0), pack-reused 25674
Receiving objects: 100% (25674/25674), 14.89 MiB | 3.30 MiB/s, done.
Resolving deltas: 100% (14498/14498), done.
Checking connectivity... done
Submodule path 'crosstool-NG': checked out '7c6bc14de33e7a331ad4932bb505385ba0963eea'
Cloning into 'esptool'...
remote: Counting objects: 164, done.
remote: Total 164 (delta 0), reused 0 (delta 0), pack-reused 164
Receiving objects: 100% (164/164), 51.49 KiB | 0 bytes/s, done.
Resolving deltas: 100% (80/80), done.
Checking connectivity... done
Submodule path 'esptool': checked out '12debb7bfe5e978060ff1919bf87d00ed52c4a72'
Cloning into 'lx106-hal'...
remote: Counting objects: 46, done.
remote: Total 46 (delta 0), reused 0 (delta 0), pack-reused 46
Unpacking objects: 100% (46/46), done.
Checking connectivity... done
Submodule path 'lx106-hal': checked out '90cdf910a8431675b6814ba447550edca3a78d4c'
make -C crosstool-NG -f ../Makefile _ct-ng
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-apple-darwin12.5.0
checking host system type... x86_64-apple-darwin12.5.0
checking for a BSD-compatible install... /opt/local/bin/ginstall -c
checking for grep that handles long lines and -e... /usr/local/bin/grep
checking for egrep... /usr/local/bin/grep -E
checking for a sed that does not truncate output... /usr/local/bin/sed
checking whether sed understands -r -i -e... yes
checking whether ln -s works... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking how to run the C preprocessor... gcc -E
checking for ranlib... ranlib
checking for gobjcopy... gobjcopy
checking for absolute path to gobjcopy... /usr/local/bin/gobjcopy
checking for gobjdump... gobjdump
checking for absolute path to gobjdump... /usr/local/bin/gobjdump
checking for greadelf... greadelf
checking for absolute path to greadelf... /usr/local/bin/greadelf
[... 128 lines cut ...]
                ^
./zconf.hash.c:193:17: error: expected expression
      {offsetof(struct kconf_id_strings_t, kconf_id_strings_str18),             T_TYPE,         TF_COMMAND, S_HEX},
                ^
./zconf.hash.c:194:17: error: expected expression
      {offsetof(struct kconf_id_strings_t, kconf_id_strings_str19),     T_OPT_DEFCONFIG_LIST,TF_OPTION},
                ^
./zconf.hash.c:196:17: error: expected expression
      {offsetof(struct kconf_id_strings_t, kconf_id_strings_str22),             T_IF,           TF_COMMAND|TF_PARAM},
                ^
./zconf.hash.c:197:17: error: expected expression
      {offsetof(struct kconf_id_strings_t, kconf_id_strings_str23),             T_TYPE,         TF_COMMAND, S_INT},
                ^
./zconf.hash.c:199:17: error: expected expression
      {offsetof(struct kconf_id_strings_t, kconf_id_strings_str27),     T_OPT_MODULES,  TF_OPTION},
                ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
1 warning and 20 errors generated.
make: *** [zconf.tab.o] Error 1
make: *** [build-lib-kconfig] Error 2
make: *** [build] Error 2
make[1]: *** [_ct-ng] Error 2
make: *** [crosstool-NG/ct-ng] Error 2
se-steowens-mb1:esp-open-sdk sowens$ 
``

moodgorning, 2015-04-21:
Hey steowens, this seems to be a mac related issue, if you paste this:

%{
# ifndef offsetof
#  include <stddef.h>
#  ifndef offsetof
#   define offsetof(st, m) ((size_t)(&((st *)0)->m))
#  endif
# endif
%}

in esp-open-sdk/crosstool-NG/kconfig/zconf.gperf 
it should fix the error. paste it right after 

%enum
%pic
%struct-type

jgarbers, 2015-04-21:
@moodgorning -- if you have a moment, would you please look at this issue #45 ?  That one feels like it might be similar to this one -- is there a similar fix? Thanks!

moodgorning, 2015-04-21:
Not sure about that one, I am still on Xcode 5.1.1 due to similar compatibility issues. No idea how to fix that particular error, sorry

jgarbers, 2015-04-21:
Okay, thanks anyway!

steowens, 2015-04-23:
Thanks jgarbers.  I have been working on a different project and just took a look at this stuff tonight.  What I did was set up the build environment in linux mint and things worked just fine.  I will try to let you know more once I am able to turn my attention to this again.  If you want to keep this open feel free to bug me until I get this done.

bhundven, 2015-11-09:
I'm curious why esp-open-sdk doesn't use the upstream crosstool-ng?
This issue has been resolved for some time.

jcmvbkbc, 2015-11-09:
@bhundven , patches that add Xtensa architecture support to crosstool-NG were posted several times to crossgcc@sourceware.org, the last one: https://sourceware.org/ml/crossgcc/2014-12/msg00100.html
The main issue raised by Yann was the way xtensa overlay was applied ( https://patchwork.ozlabs.org/patch/261562/ ). This hasn't changed.
If that's no longer an issue I'd be happy to forward-port and submit the patches again.

bhundven, 2015-11-09:
@jcmvbkbc I also do not like those changes. Seeming that there are code solutions to problems, maybe we start with the patches you have, and work through the review process to find compromise?

bhundven, 2015-11-09:
@jcmvbkbc, let me move this conversation to the previously linked issue on crosstool-ng/crosstool-ng.