pfalcon/esp-open-sdk #44: Cannot build a "blinky" example on OSX out of the box

MacDada, 2015-04-13:
```
MacBook-Pro-Dawid-2:esp-open-sdk macdada$ pwd
/Volumes/case-sensitive/esp-open-sdk

MacBook-Pro-Dawid-2:esp-open-sdk macdada$ make

cp esptool/esptool.py /Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/
ln -snf esp_iot_sdk_v1.0.0 sdk
Installing vendor SDK headers into toolchain sysroot
Installing vendor SDK libs into toolchain sysroot
Installing vendor SDK linker scripts into toolchain sysroot

Xtensa toolchain is built, to use it:
export PATH=/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:$PATH

Espressif ESP8266 SDK is installed, its libraries and headers are merged with the toolchain

MacBook-Pro-Dawid-2:esp-open-sdk macdada$ export PATH=/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:$PATH

MacBook-Pro-Dawid-2:esp-open-sdk macdada$ cd ..
MacBook-Pro-Dawid-2:case-sensitive macdada$ cd source-code-examples/blinky/

MacBook-Pro-Dawid-2:blinky macdada$ make
CC user/user_main.c
make: /opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: No such file or directory
make: *** [build/user/user_main.o] Error 1

MacBook-Pro-Dawid-2:blinky macdada$ xtensa-lx106-elf-gcc
xtensa-lx106-elf-gcc: fatal error: no input files
compilation terminated.
```

I guess I should change paths like `/opt/Espressif/` in the `blink`'s `Makefile` to simply calling the lib or to point to `/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin`?

solusipse, 2015-08-08:
I had to change two lines of `Makefile` in order to compile blinky (21: `XTENSA_TOOLS_ROOT` and 24: `SDK_BASE`:
```
20 # base directory for the compiler
21 XTENSA_TOOLS_ROOT ?= /Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin
22
23 # base directory of the ESP8266 SDK package, absolute
24 SDK_BASE        ?= /Volumes/case-sensitive/esp-open-sdk/esp_iot_sdk_v1.2.0/
```
Then it worked well.

pfalcon, 2015-08-09:
I've added recommended Makefile templates to use with esp-open-sdk:

https://github.com/pfalcon/esp-open-sdk/tree/master/examples

Using dirty, broken, or over-engineered Makefiles may lead to unexpected results (and you should contact authors of those Makefiles for support).

closed 2015-08-09 by pfalcon