pfalcon/esp-open-sdk #45: Problem building on OS X, possibly related to Xcode 6.3 (6D570)

jgarbers, 2015-04-13:
Trying to build on OS X 10.10.3 this morning after getting an Xcode update last night. Working from a clean clone of the repository and building with *make STANDALONE=n* the process fails with this error:

    [INFO ]  Installing pass-2 core C compiler
    [ERROR]    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstddef:46:9: error: no member named 'ptrdiff_t' in the global namespace
    [ERROR]    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator:402:13: error: unknown type name 'ptrdiff_t'

and many more *ptrdiff_t* related errors until the build ends with

    [ERROR]    fatal error: too many errors emitted, stopping now [-ferror-limit=]
    [ERROR]    make[4]: *** [graphite.o] Error 1
    [ERROR]    make[4]: *** [graphite-blocking.o] Error 1
    [ERROR]    make[3]: *** [all-gcc] Error 2
    [ERROR]   
    [ERROR]  >>
    [ERROR]  >>  Build failed in step 'Installing pass-2 core C compiler'
    [ERROR]  >>        called in step '(top-level)'
    [ERROR]  >>
    [ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
    [ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
    [ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
    [ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]

Not sure if this is related to the recent Xcode update or if something completely different is happening.

leonprou, 2015-04-13:
Same error here,
any solution?

hebnern, 2015-04-15:
I struggled with this all weekend as well. I finally got it working today by uninstalling Xcode and uninstalling the command line tools with:

```
cd /
lsbom -fls /var/db/receipts/com.apple.pkg.CLTools_Executables.bom | sudo xargs -I{} rm -rf "{}"
lsbom -fls /var/db/receipts/com.apple.pkg.DevSDK_OSX109.bom | sudo xargs -I{} rm -rf "{}"
lsbom -fls /var/db/receipts/com.apple.pkg.DevSDK_OSX1010.bom | sudo xargs -I{} rm -rf "{}"
sudo rm /var/db/receipts/com.apple.pkg.{CLTools_Executables,DevSDK_OSX109,DevSDK_OSX1010}.{bom,plist}
```
source: https://stackoverflow.com/questions/19635919/how-do-i-uninstall-xcode-5-command-line-tools/26456548#26456548

Then I downloaded and installed the "[Command Line Tools (OS X 10.10) for Xcode 6.1.1](developer.apple.com/downloads/download.action?path=Developer_Tools/command_line_tools_os_x_10.10_for_xcode__xcode_6.1.1/commandlinetoolsosx10.10forxcode6.1.1.dmg)" from Apple.

idserda, 2015-04-22:
Probably not the best solution, but this makes it compile:

In esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc, add this line somewhere at the top:

```#include <stddef.h>```

of the following files:

```graphite.c 
graphite-blocking.c 
graphite-clast-to-gimple.c
graphite-dependences.c 
graphite-interchange.c 
graphite-optimize-isl.c 
graphite-poly.c 
graphite-scop-detection.c 
graphite-sese-to-poly.
graphite-sese-to-poly.c
```

georgkreimer, 2015-04-24:
@idserda that worked! Thanks!

jgarbers, 2015-04-28:
Worked for me as well -- thanks, @idserda ! -- but I'm unclear about under what circumstances the changes to those files might be lost, since they're not in this repo -- I guess they get downloaded during the build process. Would *make clean* blow away those changes? I hesitate to experiment for fear of having to do them over again...

murf0, 2015-04-29:
in gmp.h at : ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in
remove __need_size_t define and undef
Source: http://stackoverflow.com/questions/17417978/compiling-gmp-library-on-osx-10-9-with-c-enabled
I did it after a failed build-attempt and a make clean x2
TNTKY!

jgarbers, 2015-04-29:
@murf0 are you saying that fix works *without* making the changes to the graphite files @idserda mentioned?

murf0, 2015-04-29:
@jgarbers Yes, sorry was a bit unclear in the description. Basically the issue is with GMP library on the latest Xcode. This change forces the import of ptrdiff_t. Making it compile. (As far as i can tell..)

Tylerflick, 2015-05-11:
@sebgiles The `.build` directory is created when you run `make`.

murf0, 2015-05-11:
You need to have the make fail first i believe to have the .build directory created (and download all required tar-balls)

sebgiles, 2015-05-12:
@Tylerflick Thanks I'll try it now!
Intuitively I deleted the whole thing and downloaded it again, it feels odd that I can't fix the build error before it actually occurs.

murf0, 2015-05-12:
That's because the first build-process downloads and kickstarts the secondary build-processes needed. So before the first try there is no source to build, the sources are downloaded.

pacovar, 2015-05-23:
**@murf0** Beauty, thanks.

anusoft, 2015-05-26:
from @idserda comment I've added command line for lazy people like me

```
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-blocking.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-clast-to-gimple.c
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-dependences.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-interchange.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-optimize-isl.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-poly.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-scop-detection.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-sese-to-poly.c
```

Anton-Rodionov, 2015-06-08:
@anusoft Thanks, it works!

jiegec, 2015-07-12:
@pfalcon I think we should include the solution from @anusoft in the README.md for Mac OS X users.

pfalcon, 2015-07-12:
Any crosstool-NG patches should go to crosstool-NG project.

Palantir555, 2015-08-14:
Thanks, @murf0! That worked like a charm.

For reference, this is the sed command I use to fix the issue using murf0's method: `sed -i '/__need_size_t/d' ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in`

whmountains, 2015-08-25:
@Palantir555 Thanks so much.  Your one-liner fixed it for me.

joostn, 2015-09-16:
Somehow that sed command failed when I tried, but this worked for me:
`sed -i.bak '/__need_size_t/d' ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in `

austimbo, 2015-09-22:
A big thank you for this post. Commenting out "__need_size_t define and undef" did the job for me. I was going to be lost without it.

zdrc, 2015-11-09:
I'm trying to build on OSX 10.11.1 with Xcode 7.1 and I'm running into an identical problem but these fixes aren't working.

First of all, "__need_size_t define and undef" is not in ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in

What is in ./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in is
```
#define __need_size_t  /* tell gcc stddef.h we only want size_t */
#if defined (__cplusplus)
#include <cstddef>     /* for size_t */
#else
#include <stddef.h>    /* for size_t */
#endif
#undef __need_size_t
```

I've removed this section of code, ran make clean 2x, then make again to no avail. This code always reappears in gmp-h.in after running make. I've also tried the sed commands of @anusoft, and that hasn't worked either. Any suggestions?

Palantir555, 2015-11-09:
@zdrc You're not supposed to remove that whole block of code, but just the first and last lines (`#define __need_size_t /*...*/` and `#undef __need_size_t`). Try going back to the original contents of the file and removing those 2 lines, and let us know if that fixes your issue.

zdrc, 2015-11-09:
I just deleted `#define __need_size_t` and `#undef __need_size_t`, ran make clean 2 x, and attempted to build again. It failed with the same error message.

Palantir555, 2015-11-10:
@zdrc the file you are modifying is downloaded during the make process (before erroring out).

You should run the sed command I posted and run `make` again right away. Do NOT run `make clean` after seeing the error.

By running `make clean` you are removing the file you just modified, then when you run `make` you download the original version again and see the same error because you've got the same file.

Hope that helps :)

zdrc, 2015-11-10:
@Palantir555 OK, I'll give that a go. Unfortunately, it seems like http://www.mpfr.org is down today, so I can't build at the moment.

recursify, 2015-11-11:
@zdrc Yep `mpfr` and `isl` packages are unreachable.

Download these alternate links into
`esp-open-sdk/crosstool-NG/.build/tarballs`

http://pkgs.fedoraproject.org/repo/pkgs/gcc/isl-0.12.2.tar.bz2/e039bfcfb6c2ab039b8ee69bf883e824/isl-0.12.2.tar.bz2
https://ftp.gnu.org/gnu/mpfr/mpfr-3.1.2.tar.xz

zdrc, 2015-11-11:
@Palantir555 I did as you suggested and it built.

@recursify Earlier today I changed the shell script that was fetching `mpfr` to get the package from http://ftp.gnu.org/gnu/mpfr/ instead. I didn't have a problem with `isl` while building.

Thanks all

kalanda, 2015-12-04:
Both solutions (@anusoft and @murf0) did work for me using OSX 10.11.1 (15B42). 
Thanks

eyp, 2015-12-09:
Xcode 7.1.1 on OSX 10.11.1 works like a charm with the @murf0 patch `doing make afterwards. Remember, don't make a clean after patching.

Tkizzy, 2016-03-13:
@Tylerflick is trolling. 
```
Pull the repo.
Execute make. (You'll see the above error)
Perform the edit listed by @murf0.
Execute make clean && make.
```

DO NOT DO A MAKE CLEAN AFTER THE EDIT! IT WILL ERASE THE EDIT AND YOU'LL HAVE TO DO IT AGAIN.

instead just make after doing the edit.

jacobjennings, 2016-03-14:
if you don't have gsed,
brew install gnu-sed
ln -s /usr/local/Cellar/gnu-sed/4.2.2/bin/sed ~/bin/gsed
assuming ~/bin is already in your path.

mcauser, 2016-05-05:
The @murf0 patch works on Xcode 7.2 (7C68) on OSX 10.10.5 Yosemite.
./crosstool-NG/.build/src/gmp-5.1.3/gmp-h.in
`//#define __need_size_t`
`//#undef __need_size_t`