pfalcon/esp-open-sdk #49: Compiling Error: Build failed in step 'Installing final compiler'

s137, 2015-05-02:
I tried building the esp-open-sdk in order to compile nodemcu for my ESP.

I got the following error on my Ubuntu 10.04.4 32bit System, during the building process:

[INFO ]  Installing final compiler
[ERROR]    make[4]: *** [cc1] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing final compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_for_host[scripts/build/cc/gcc.sh@595]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 53:36.00)
[53:37] / make[2]: *** [build] Fehler 2
make[2]: Verlasse Verzeichnis '/home/simon/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Fehler 2
make[1]: Verlasse Verzeichnis '/home/simon/esp-open-sdk/crosstool-NG'
make: *** [/home/simon/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Fehler 2

I hope you can help me with this.
Thanks in Advance.
s137

pfalcon, 2015-05-03:
See #30 (and search existing reports first).

s137, 2015-05-03:
Alright, thanks. I created an issue at 'jcmvbkbc/crosstool-NG'.

closed 2015-05-03 by s137