pfalcon/esp-open-sdk #50: patch: adding Espressif libnet80211.a patch to

HappyCodingRobot, 2015-05-14:
Makefile patch to support Espressif's libnet fix
(http://bbs.espressif.com/viewtopic.php?f=5&t=426&sid=c6355a0f51f6d5142f6e7fd8329bc0ed)

```patch
--- Makefile	2015-04-26 14:32:12.886110295 +0200
+++ Makefile.patch_v101	2015-05-14 15:51:27.891186061 +0200
@@ -58,7 +58,9 @@

 sdk_patch: .sdk_patch_$(VENDOR_SDK)
 
-.sdk_patch_1.0.1:
+.sdk_patch_1.0.1: libnet80211.zip esp_iot_sdk_v1.0.1/.dir
+	$(UNZIP) $<
+	mv libnet80211.a $(VENDOR_SDK_DIR_1.0.1)/lib/
 	patch -d $(VENDOR_SDK_DIR_1.0.1) -p1 < c_types-c99.patch
 	@touch $@
 
@@ -119,6 +121,8 @@
 	wget --content-disposition "http://bbs.espressif.com/download/file.php?id=190"
 libssl.zip:
 	wget --content-disposition "http://bbs.espressif.com/download/file.php?id=316"
+libnet80211.zip:
+	wget --content-disposition "http://bbs.espressif.com/download/file.php?id=361"
 
 sdk: $(VENDOR_SDK_DIR)/.dir
 	ln -snf $(VENDOR_SDK_DIR) sdk
```

pfalcon, 2015-05-19:
Why not submit proper pull request?

HappyCodingRobot, 2015-05-20:
Sorry :/
next time the proper way..

closed 2015-05-20 by HappyCodingRobot