pfalcon/esp-open-sdk #53: "Blinky" example fails to build with SDK version 1.1.0

m-thu, 2015-05-25:
Compiling the "Blinky" example with SDK version 1.1.0 (Ubuntu  14.04.2 LTS) fails with the following error message:
```
xxx@xxx:/opt/Espressif/source-code-examples/blinky$ make
CC user/user_main.c
AR build/app_app.a
LD build/app.out
/opt/Espressif/ESP8266_SDK/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x34c): undefined reference to `user_rf_pre_init'
/opt/Espressif/ESP8266_SDK/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x474): undefined reference to `user_rf_pre_init'
collect2: error: ld returned 1 exit status
make: *** [build/app.out] Error 1
```

If I compile the toolchain with the VENDOR_SDK variable in the Makefile set to 1.0.1, the example builds just fine.

pprimm, 2015-05-26:
Lubuntu VM & Ubuntu 15.04 same error for SDK v 1.1.0.  Similar errors in all other projects.  Reverted back to v 1.0.1, added patch for station+softAP issue (http://bbs.espressif.com/viewtopic.php?f=5&t=426) and build works again.

pfalcon, 2015-05-26:
If you read 1.1.0 release notes http://bbs.espressif.com/viewtopic.php?f=5&t=481 ,  you'll see "9.user_rf_pre_init has to be added in user_main.c, user can call system_phy_set_rfoption to set configuration of RF in it."

Yes, that's not the highest quality engineering on the vendor part. Incompatibility could have been easily avoided. But today 1.1.0 was re-released under MIT license, which makes it first valid SDK release (in a sense as not putting unrealizable liability on users), so esp-open-sdk stays on 1.1.0.

The issue above can still be patched - pull request welcome, or maybe I'll get to it. In the meantime, update your source.

pfalcon, 2015-05-26:
Fixed in e03adb67c03216d13db26b9101edf9e3344f5b2c. Note that there're other backward-incompatible changes in 1.1.0. ESP8266 SDK only now starts to stabilize, everyone who used ESP8266 before was a beta-tester and of course knew that.

pfalcon, 2015-05-26:
Actually, fixed in 4e4590de89bc1524f6b7e7c35744325f9dc2e27b.

closed 2015-05-26 by pfalcon