pfalcon/esp-open-sdk #58: Ubuntu 15.10 requires libtool-bin

heeen, 2015-06-11:
checking for GNU libtool >= 1.5.26... no
configure: error: could not find GNU libtool >= 1.5.26

thehans, 2015-06-17:
I have this same error while attempting to build on Ubuntu 15.04.  

    checking for GNU libtool >= 1.5.26... no
    configure: error: could not find GNU libtool >= 1.5.26
    ../Makefile:238: recipe for target '_ct-ng' failed
    make[1]: *** [_ct-ng] Error 1
    make[1]: Leaving directory '/opt/esp-open-sdk/crosstool-NG'
    Makefile:235: recipe for target 'crosstool-NG/ct-ng' failed
    make: *** [crosstool-NG/ct-ng] Error 2

I have libtool 2.4.2-1.11 installed.

pfalcon, 2015-07-03:
Addressed in https://github.com/pfalcon/esp-open-sdk/commit/319c54520ca3aeb0b0bdab628a96b3ce51f47eef

closed 2015-07-03 by pfalcon

cebrax, 2015-10-28:
Installing libtool-bin solves this issue.

mindforger, 2019-04-08:
why is the hint from cebrax not added to the readme?