pfalcon/esp-open-sdk #60: Documentation - wrong $PATH recommendation

sej7278, 2015-06-24:
This is a really bad idea (security, usability etc.) the toolchain should be appended to the end of the existing $PATH, not inserted at the start of it.

https://github.com/pfalcon/esp-open-sdk/blob/master/Makefile#L45

```
Xtensa toolchain is built, to use it:

export PATH=/home/user/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
```

pfalcon, 2015-06-24:
ORLY? And what will you tell folks who have "xtensa-lx106-elf-gcc" trojan horse in their PATH, so they send reports "I built your piece, and when I run it, I hear only snickers"? I personally won't tell them anything, because I already **have told** them to put it at the beginning of their PATH.

closed 2015-06-24 by pfalcon

sej7278, 2015-06-24:
i think you're getting your logic backwards.

if you have the xtensa stuff at the beginning of the $PATH, then any binaries in there will take precedence over the ones in your $PATH.

so a trojan called "ls" in the xtensa directory will override /bin/ls

pfalcon, 2015-06-24:
No, *you* try to miss obvious things - everyone who adds any directory to their PATH first checks that it doesn't contain ls and stuff (except for everyone who doesn't of course, but it's their choice). Then the only way "ls" can be in bin dir built by esp-open-sdk is that: you have your system trojaned already - your choice, makes no sense to discuss it. Otherwise, either esp-open-sdk, @jcmvbkbc's fork of crosstool-NG, crosstool-NG upstream, or FSF gcc ship you trojans. You should review what they ship to you before using it. Fortunately, all these parties give you source code, so it's possible to review it, and least-known parties like me and @jcmvbkbc give you little enough source code so it's even easy to review it.

OTOH, there're bunch of people who ship you binaries of unknown origin. It's hard to assess what's inside them, and impossible (or makes no sense) to assess whether user has something like that or not (we're not government to poke nose into your PATH). The only reliable choice is to structure it the way that it doesn't depend on whether user has it or not. And that both reliable and obvious choice is to put it at the front of the PATH.

All the above is so obvious that I'm really embarrassed to write it [again, for the Nth time].

sej7278, 2015-06-24:
if you're ok advocating putting a user-writable directory at the beginning of the search path, that's up to you. i'm not going to waste my time trying to dissuade you.