pfalcon/esp-open-sdk #69: lwip build from source?

eastein, 2015-07-21:
Currently liblwip.a is being taken direct from the espressif SDK binary libs, correct? Have you explored the work done by @kadamski at https://github.com/kadamski/esp-lwip - is there a reason not to use this?

I suppose since liblwip source is not being regularly released by Espressif this might be a problem - this was based on a dump of changes on top of 1.4.0.

pfalcon, 2015-07-21:
Yes, sure, doing it like that was the idea behind esp-open-sdk, except that I cannot do it all myself. @kadamski's work is nice and good progress towards that aim, but:

1. His project isn't up to date wrt to recent SDKs - not only latest 1.2.0, but even 1.0.1: https://github.com/kadamski/esp-lwip/issues/7
2. @kadamski himself isn't very responsive.

I opened https://github.com/kadamski/esp-lwip/issues/9 trying to discuss how to ease maintenance of his lwIP port, but that all appears to be blocked on time availability.

Summing up: if you have time/interest, feel free to start on that and keep in touch.

eastein, 2015-07-22:
Yeah, I kind of figured this would be the reason. I wanted to make sure there wasn't a "Well, this is never going to work well for x reasons" reason to not do it before I considered if I wanted to do it.

eastein, 2015-07-22:
It seems to be a pretty huge task to make that work for all SDK versions, or even any recent ones - and the amount of testing it has received has so far been low.

eastein, 2015-07-22:
Thanks for the info; perhaps at some point I will have the energy for it.

pfalcon, 2015-07-22:
> "Well, this is never going to work well for x reasons"

Nope, main concerns are "it would be nice, and can be done, but I don't need it that much (I'm satisfied with 'can be done' assessment I'm performed)" and "esp8266 is just a random chip, popping out of nowhere several months ago and likely going back in some time". Note that various people doing these things in different ways, there's for example https://github.com/SuperHouse/esp-open-rtos , which is on a way to build lwIP, axTLS, etc. from source - but in FreeRTOS environment.

eastein, 2015-07-22:
Ah nice, thanks for the esp-open-rtos pointer.

stv0g, 2015-09-07:
Espresso recently released their version of LwIP in source format:
http://bbs.espressif.com/viewtopic.php?f=46&t=951

This should allow us to build it at least for the newest (1.3.0) SDK version.
I stil have to test, weather their own SDK properly links against a manually compiled LwIP.

eastein, 2015-09-07:
oooh nice

eastein, 2015-09-07:
I care about this because the built in stack only supports A records for DNS, and I want SRV. So I've been implementing my own SRV queryer with a little bit of code taken from LWIP. I'd like to contribute it back possibly by integrating it in, although I'm not sure the LWIP folks care about it.

jrmuizel, 2016-06-22:
Is this resolved by https://github.com/pfalcon/esp-open-lwip?

pfalcon, 2017-05-21:
Yes, https://github.com/pfalcon/esp-open-lwip largely resolved this. And the final step is https://github.com/pfalcon/esp-open-sdk/issues/263

closed 2017-05-21 by pfalcon