pfalcon/esp-open-sdk #76: cloog download is missing

civic95man, 2015-08-03:
The build fails when trying to download cloog. Tried `make clean` and rebuilding but it can't find it.

```
[DEBUG]    Trying 'http://www.bastoul.net/cloog/pages/download/cloog-0.18.1'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/parallels/Desktop/Espressif/esp-open-sdk/crosstool-NG/.build/tarballs/cloog-0.18.1.tmp-dl' 'http://www.bastoul.net/cloog/pages/download/cloog-0.18.1' 
[ALL  ]    --2015-08-03 16:23:27--  http://www.bastoul.net/cloog/pages/download/cloog-0.18.1
[ALL  ]    Resolving www.bastoul.net (www.bastoul.net)... failed: Connection timed out.
[ALL  ]    wget: unable to resolve host address `www.bastoul.net'
[DEBUG]    Not at this location: "http://www.bastoul.net/cloog/pages/download/cloog-0.18.1"
[DEBUG]    Trying 'ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.1'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/parallels/Desktop/Espressif/esp-open-sdk/crosstool-NG/.build/tarballs/cloog-0.18.1.tmp-dl' 'ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.1' 
[ALL  ]    --2015-08-03 16:23:37--  ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.1
[ALL  ]               => `/home/parallels/Desktop/Espressif/esp-open-sdk/crosstool-NG/.build/tarballs/cloog-0.18.1.tmp-dl'
[ALL  ]    Resolving gcc.gnu.org (gcc.gnu.org)... failed: Connection timed out.
[ALL  ]    wget: unable to resolve host address `gcc.gnu.org'
[DEBUG]    Not at this location: "ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.1"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_cloog_get[scripts/build/companion_libs/130-cloog.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@583]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 2:41.38)
```

Seems to not exist anymore….  Any help is appreciated. Thanks

jcmvbkbc, 2015-08-04:
The following lines from the log suggest that you may have a problem with DNS or internet connection:
```
[ALL  ]    Resolving www.bastoul.net (www.bastoul.net)... failed: Connection timed out.
[ALL  ]    wget: unable to resolve host address `www.bastoul.net'
...
[ALL  ]    Resolving gcc.gnu.org (gcc.gnu.org)... failed: Connection timed out.
[ALL  ]    wget: unable to resolve host address `gcc.gnu.org'
```
It works well on my side:
```
[DEBUG]    Trying 'http://www.bastoul.net/cloog/pages/download/cloog-0.18.1.tar.gz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/jcmvbkbc/tmp/esp/opt/Espressif/tmp/crosstool-NG/.build/tarballs/cloog-0.18.1.tar.gz.tmp-dl' 'http://www.bastoul.net/cloog/pages/download/cloog-0.18.1.tar.gz'
[ALL  ]    --2015-08-04 03:04:12--  http://www.bastoul.net/cloog/pages/download/cloog-0.18.1.tar.gz
[ALL  ]    Resolving www.bastoul.net (www.bastoul.net)... 62.210.16.62
[ALL  ]    Connecting to www.bastoul.net (www.bastoul.net)|62.210.16.62|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 200 OK
[ALL  ]    Length: 3857324 (3.7M) [application/x-gzip]
[ALL  ]    Saving to: '/home/jcmvbkbc/tmp/esp/opt/Espressif/tmp/crosstool-NG/.build/tarballs/cloog-0.18.1.tar.gz.tmp-dl'
[ALL  ]
[ALL  ]         0K ................ ................ ................ 10% 1.18M 3s
[ALL  ]       384K ................ ................ ................ 20% 2.17M 2s
[ALL  ]       768K ................ ................ ................ 30% 2.25M 1s
[ALL  ]      1152K ................ ................ ................ 40% 2.14M 1s
[ALL  ]      1536K ................ ................ ................ 50% 2.16M 1s
[ALL  ]      1920K ................ ................ ................ 61% 2.39M 1s
[ALL  ]      2304K ................ ................ ................ 71% 2.26M 1s
[ALL  ]      2688K ................ ................ ................ 81% 2.24M 0s
[ALL  ]      3072K ................ ................ ................ 91% 2.17M 0s
[ALL  ]      3456K ................ ................ ......          100% 2.25M=1.8s
[ALL  ]
[ALL  ]    2015-08-04 03:04:14 (2.04 MB/s) - '/home/jcmvbkbc/tmp/esp/opt/Espressif/tmp/crosstool-NG/.build/tarballs/cloog-0.18.1.tar.gz.tmp-dl' saved [3857324/3857324]
[ALL  ]
[DEBUG]    Got it from: "http://www.bastoul.net/cloog/pages/download/cloog-0.18.1.tar.gz"
[DEBUG]    Got 'cloog-0.18.1' from the Internet
```

civic95man, 2015-08-04:
hmmm, okay. I know it's able to download gmp, isl, and mpfr but it always fails at cloog. I'll have to investigate further. thanks

closed 2015-08-04 by pfalcon