pfalcon/esp-open-sdk #78: Can't get IoT demo example to build..

DrObscure, 2015-08-06:
Trying to build the default IoT example in the sdk/examples tree.. should these be buildable as delivered, or did I miss reading a document, etc..??

All I get from the make when running gen_misc.sh is this after the input questions, which I choose the defaults:
----------------------------------------------------
start...

gmake: Nothing to be done for 'FORCE'.
---------------------------------------------------------
The makefile seems rather sparse, an I am under the impression there is no parent makefile that should be called above it..

DrObscure, 2015-08-08:
Ok.. rewrote the makefile to work locally...

closed 2015-08-08 by DrObscure

g-goessel, 2015-08-08:
Can you share your solution?

DrObscure, 2015-08-10:
This message was originally HTML formatted.  View in a HTML capable client to see the original version.\r\n\r\nBe glad to once I resolve a linking issue I'm having..I'll post it when completed..---- Original Message ----From: "g_goessel" notifications@github.comSent: 8/8/2015 1:44:00 AMTo: "pfalcon/esp-open-sdk" esp-open-sdk@noreply.github.comCc: "Dr. Obscure" chane@carrollsweb.comSubject: Re: [esp-open-sdk] Can't get IoT demo example to build.. (#78)Can you share your solution? Le 8 aot 2015 8:38 AM, "Dr. Obscure" notifications@github.com a crit :Ok.. rewrote the makefile to work locally...

l29ah, 2016-03-14:
Any luck?

DrObscure, 2016-03-15:
Hi Sergey.. yes i did finally :)
I did have issues trying to load the firmware to the device, and haven't 
had time revisit it again - sigh..
I hope to get back into it again.. its on my schedule for next month.. :)

rakeshsafir, 2016-04-09:
Hey DrObscure.
I'm troubled with the same issue of examples not compiling. The issue is same with every example in the sdk. Seems you solved the compilation part. Can you please share your solution.

DrObscure, 2016-04-09:
This message was originally HTML formatted.  View in a HTML capable client to see the original version.\r\n\r\nHi Rakesh...Its been a while since I did this, and will need to revisit my notes and such..This next coming week I am traveling, so it will have to wait until I get back to my office and such..FYI, I did this on a FreeBSD 10.1 system, so it may be different on another systemTalk at you after next week or so..  --DrObscure---- Original Message ----From: "Rakesh Ranjan" notifications@github.comSent: 4/9/2016 2:38:02 AMTo: "pfalcon/esp-open-sdk" esp-open-sdk@noreply.github.comCc: "Dr. Obscure" chane@carrollsweb.comSubject: Re: [pfalcon/esp-open-sdk] Can't get IoT demo example to build.. (#78)Hey DrObscure.I'm troubled with the same issue of examples not compiling. The issue is same with every example in the sdk. Seems you solved the compilation part. Can you please share your solution.