pfalcon/esp-open-sdk #85: SDK Makefile versus root Makefile problem

jfbortolotti, 2015-08-29:
While willing to compile an examples in sdk/app, got the following error:
~/esp8266/esp-open-sdk/sdk/app$ make
../../Makefile:288: warning: overriding commands for target `clean'
../Makefile:299: warning: ignoring old commands for target `clean'
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
make -C crosstool-NG -f ../Makefile _ct-ng
make: *** crosstool-NG: No such file or directory.  Stop.
make: *** [crosstool-NG/ct-ng] Error 2

This seems to be due to the app Makefile including the ../../Makefile which is the one from root but with a different path, then trying to regenerate the crosstool-NG compile/install process.

Workaround found: rename esp-open-sdk/Makefile to something else(Makefile.old) and app compilation will work.

me21, 2015-09-04:
Or patch Makefile in the root of SDK to prevent it from including Makefile from parent dir.

pfalcon, 2015-09-04:
Or don't do anything is esp-open-sdk source tree, because it's all, well, esp-open-sdk source tree. Clone vendor sdk outside the tree and play with it there. Or report the issue to vendor, telling them not to include random makefiles outside their tree.

closed 2015-09-04 by pfalcon

eku, 2015-09-04:
Well, I had the same problem and renaming the files outside the vendor tree from Makefile to makefile did the trick.

Scott--R, 2016-05-27:
Same issue for me. I spent 2 days frustrated by this error, and then stumbled onto this thread. No matter where I put the source code I always got this exact error. But the moment I renamed the Makefile in the root directory, it compiled perfectly. I will simply change the Makefile name back to normal to pull updates, but I am so glad I finally got it working.

surak, 2018-01-11:
This error is caused by the BSD's "sed" utility. Try "brew install gnu-sed binutils"

surak, 2018-01-11:
And wget, if you don't have it.