pfalcon/esp-open-sdk #86: Cannot Build Toolchain

kratz9, 2015-09-10:
Using Lubunto, latest version 15.04 32bit, fresh VM. Failed after following directions on the Readme.

First error i recieved was this:

checking for GNU libtool >= 1.5.26... no
configure: error: could not find GNU libtool >= 1.5.26
../Makefile:277: recipe for target '_ct-ng' failed
make[1]: *** [_ct-ng] Error 1

The libexpat-dev dependency was already downloaded. Following advice found elsewhere, i did apt-get install libtool-bin. That got me past that error. But then while trying to do 'make STANDALONE=n', I recieved another error. 

make[2]: Entering directory '/home/kratz/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20150909.203422
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 216.05s (at 03:37)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 105.15s (at 05:22)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 76.44s (at 06:39)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 38.04s (at 07:17)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 68.31s (at 08:25)
[INFO ]  =================================================================
[INFO ]  Installing CLooG for host
[INFO ]  Installing CLooG for host: done in 11.35s (at 08:37)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 11.96s (at 08:49)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 144.77s (at 11:13)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C compiler
[ERROR]    collect2: error: ld returned 1 exit status
[ERROR]    make[4]: *** [cpp] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 21:07.45)
[21:07] / ct-ng:148: recipe for target 'build' failed
make[2]: *** [build] Error 2

What do I do from here?

jcmvbkbc, 2015-09-10:
Please share your build.log file.

kratz9, 2015-09-10:
Here is the last bit from the build.log. It's 10MB and the site has no option to attach files.

[ALL  ]    /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/builtins.c: In function 'void expand_builtin_return(rtx)':
[ALL  ]    /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/config/xtensa/xtensa.h:378:41: warning: comparison of unsigned expression >= 0 is always true [-Wtype-limits]
[ALL  ]          ((unsigned) ((OUT) - GP_REG_FIRST) >= WINDOW_SIZE)) ?                                           ^
[ALL  ]    /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/builtins.c:1707:27: note: in expansion of macro 'INCOMING_REGNO'
[ALL  ]      reg = gen_rtx_REG (mode, INCOMING_REGNO (regno));
[ALL  ]                               ^
[ALL  ]    i686-build_pc-linux-gnu-g++ -c   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libbacktrace -DCLOOG_INT_GMP -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/calls.c -o calls.o
[ALL  ]    i686-build_pc-linux-gnu-g++ -c   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libbacktrace -DCLOOG_INT_GMP -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/combine.c -o combine.o
[ALL  ]    i686-build_pc-linux-gnu-g++ -c   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libbacktrace -DCLOOG_INT_GMP -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/kratz/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  /home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/dojump.c -o dojump.o
[ALL  ]    i686-build_pc-linux-gnu-g++ -c   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/. -I/home/kratz/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../include [... cut ...]
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 21:07.45)

jcmvbkbc, 2015-09-10:
Looks like insufficient memory size. Please make sure you have at least 1G free RAM+swap on your build machine.

kratz9, 2015-09-10:
Yep that worked. I had used the defaults for the VM and the install which were 512 RAM and 512 Swap. I increased to 1G ram and 1G Swap.

closed 2015-09-10 by kratz9