pfalcon/esp-open-sdk #89: Unable to build on Ubunutu 14.04

rubberbird, 2015-09-18:
esp-open-sdk/crosstool-NG/.build/tarballs/mpc-1.0.2.tar.gz: not in gzip format

pfalcon, 2015-09-18:
Random download error on your side.

closed 2015-09-18 by pfalcon

rubberbird, 2015-09-18:
Thats what I thought, but after downloading the file seperately for 4 different sources and testing a tar xvf manually to make sure it was the right format, still the same error. Do you have a good source of this file that I can test?

rubberbird, 2015-09-18:
Pulling from the official site and it works. Sorry about that. Thanks for the code :+1:

Top-Dog, 2016-01-14:
What were the steps you used to get around this? I can't seem to get around this problem.

SBlincov, 2018-02-14:
@Top-Dog Try to do it:
1) Delete crosstool-NG from root of esp-open-sdk
2) Make tmp directory
`mkdir tmp`
`cd tmp`
3) Clone crosstool-NG repository 
`git clone https://github.com/crosstool-ng/crosstool-ng`  
4) Replace downloaded crosstool to root of esp-open-sdk
5) Change name "crosstool-ng" on "crosstool-NG"