pfalcon/esp-open-sdk #90: Missing sint16_t on c_types.h

amirgon, 2015-09-18:
With SDK 1.3.0, `c_types.h` includes `<stdint.h>`, and both are missing `sint16_t`. (appears only in `#if 0`)  

I Had to add a typedef for `sint16_t` in c_types.h in order to build [lwip_open_src_template_proj_for_v1.3.0](http://bbs.espressif.com/viewtopic.php?f=46&t=951).  

In my opinion - worth adding a patch.

pfalcon, 2015-10-07:
There're few points:

1. "sint16_t" is not a standard type (int16_t is), so it it's groundedly not in <stdint.h>.
2. Using non-standard types, especially if there's corresponding standard one, is ungood.
3. If you're working on building lwip_open_src_template_proj_for_v1.3.0, please consider contributing patches to esp-open-sdk, I'd love to see open-source lwip build in it (but stuff in that tarball would certainly need clean-up).

amirgon, 2015-10-11:
Ok.
Added pull request with lwip_open_src_template.

pfalcon, 2016-01-07:
This is fixed in the latest master.

closed 2016-01-07 by pfalcon