pfalcon/esp-open-sdk #93: $ make STANDALONE=y build

lloyd-g, 2015-09-23:
followed "$ make STANDALONE=y" but  xtensa-lx106-elf-gcc was not installed into opt/ folder.
and when I run make file pointing to ~/esp-open-sdk/ and folders below it can't find ld scripts but they are there?
is there a good makefile and example you could recommended?

systemparanoia, 2015-12-31:
Also having issues with the same.
doesnt work if i run it as sudo or not

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/home/crunch/GIT/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/home/crunch/GIT/esp-open-sdk/crosstool-NG'
[ERROR]  You must NOT be root to run crosstool-NG
[00:00] / ct-ng:148: recipe for target 'build' failed
make[2]: *** [build] Error 1
make[2]: Leaving directory '/home/crunch/GIT/esp-open-sdk/crosstool-NG'
../Makefile:282: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/crunch/GIT/esp-open-sdk/crosstool-NG'
Makefile:279: recipe for target '/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

jmccutchan, 2016-05-19:
same issue:make[2]: Entering directory `/root/esp-open-sdk/crosstool-NG'
[ERROR]  You must NOT be root to run crosstool-NG
[00:00] / make[2]: *** [build] Error 1
make[2]: Leaving directory `/root/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/root/esp-open-sdk/crosstool-NG'
make: *** [/root/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

Anwaarullah, 2016-09-20:
Give your normal user full read/write/execute permissions on the esp-open-sdk folder. Eg: [http://serverfault.com/questions/105535/how-can-i-set-full-premissions-to-a-user-in-a-specified-dir](http://serverfault.com/questions/105535/how-can-i-set-full-premissions-to-a-user-in-a-specified-dir)

Then, simply use **make** instead of **sudo make**

chriva, 2017-11-26:
Just enter crosstool-NG/scripts and edit crosstool-NG.sh.in.
Which account you use is up to you and not something a tinnfoil-hat wearing buffoon should decide.

kaizoku-619, 2018-06-06:
@chriva thanks