pfalcon/esp-open-sdk #94: OSX El Capitan error

eliabieri, 2015-09-26:
I tried to set the SDK on OS X El Capitan and I got the following error:

[ERROR]  >>  Build failed in step 'Installing pass-2 core C compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@483]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 16:35.00)
[16:35] / make[2]: *** [build] Error 1
make[1]: *** [_toolchain] Error 2

impressiver, 2015-10-03:
@eliabieri there's not much to go on w/ just the tail of the log, but as a random suggestion, this might help (paste this it into a file at the root the repo, name it something like `osx-bandaid.sh`): 
```.sh
#!/usr/bin/env bash
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-blocking.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-clast-to-gimple.c
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-dependences.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-interchange.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-optimize-isl.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-poly.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-scop-detection.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-sese-to-poly.c
```
I swiped this list from *somebody* awesome in one of these threads, and I use it after pretty much any considerable upstream change. Search the issues and you'll find it's origin story, but here are some pointers:
```.sh
$ make clean all              # the build will fail
$ ./osx-bandaid.sh            # magical sed replacements
$ make STANDALONE=no          # this time the build will succeed, as long as you *believe*
```
If you're not so lucky, don't give up. It didn't work for me the first time I followed someone else's instructions either.

Do yourself a favor: spin up a VM/container to use as a dedicated esp-dev environment. With all the packages to install (`coreutils`, `gnu*`, etc), environment vars, `$PATH`s, and case-sensitive filesystem, flashing chips is way more fun than setting up toolchains... and I've spent more hours than I care to admit resurrecting my esp toolchain after system upgrades, not seeing eye to eye with Rust/Go/Python's philosophies, or accidentally running `brew upgrade --all` without paying attention.

Now I have everything in a VM, and life is better.

eliabieri, 2015-10-13:
Thank you very much for the help!

closed 2015-10-13 by eliabieri