pfalcon/esp-open-sdk #98: TARGET_ALIAS undeclared

con-f-use, 2015-10-05:
I get an "esp-open-sdk/crosstool-NG/.build/src/binutils-2.24/gas/as.c:224:14: error: 'TARGET_ALIAS' undeclared (first use in this function)" now with the [latest commit](https://github.com/pfalcon/esp-open-sdk/commit/b64eebdbb0c8bcbd3961fd2751653a1ae87a559b) on Ubuntu 14.04.3 LTS. Tried it on two different systems, one of them fresh. What is the correct value for this define? Also there are some other undefined values. See the log:

>  Build failed in step 'Installing binutils for host'
>  Error happened in: CT_DoExecLog[scripts/functions@257]
[build.log](http://paste.ubuntu.com/12553731/)

con-f-use, 2015-11-09:
Is it dead jim?

pfalcon, 2015-11-09:
I'm using Ubuntu Trusty myself, cannot reproduce. Otherwise, see https://github.com/pfalcon/esp-open-sdk/issues/101 .

closed 2015-11-09 by pfalcon

con-f-use, 2017-02-07:
I found that it was something related to Ubuntu's shell settings after installing a certain package (not sure which one). Running make from a clean shell `env -i bash --norc --noprofile` solved the problem.