pfalcon/esp-open-sdk #103: C++ issue with std::map

mbtronics, 2015-10-19:
While trying to use an std::map I got the following errors:

    std::map<int, int> MyMap;
    MyMap[5] = 6;

esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_tree.h:1130: undefined reference to `std::_Rb_tree_decrement(std::_Rb_tree_node_base*)'

And several others related to the functions in esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/libstdc++-v3/src/c++98/tree.cc. I tried several flags and other things, without result. Now I just add tree.cc and all necessary includes to my project.

pfalcon, 2015-10-19:
> I tried several flags and other things, without result.

I don't know what response people expect with reports like that. Something like "just try that other flag" ?. That's why there's https://github.com/pfalcon/esp-open-sdk/issues/101 - to at least make people report issues where they belong as first step towards reporting issues right.

closed 2015-10-19 by pfalcon

mbtronics, 2015-10-20:
With all due respect, the reason I created the issue here is because I
suspect the makefile has something to do with it.
I compiled the toolchain manually with crosstool-ng and then it did work,
but I don't have a clue what the difference could be.

pfalcon, 2015-10-20:
> I compiled the toolchain manually with crosstool-ng and then it did work,
but I don't have a clue what the difference could be.

But you didn't tell that. And you still don't tell 95% of the information which is needed to get use of a bug report. Google for "how to report bugs". Random link from 1st page: http://www.softwaretestinghelp.com/how-to-write-good-bug-report/ . Random quote from this random link:

> 2) Reproducible:
If your bug is not reproducible it will never get fixed. You should clearly mention the steps to reproduce the bug. Do not assume or skip any reproducing step. Step by step described bug problem is easy to reproduce and fix.