pfalcon/esp-open-sdk #104: how to configure build process to build gdb --with-python

rabarar, 2015-10-30:
I'd like to generate a python-enabled version of gdb, but I'm uncertain how and where to add the --with-python to the configuration call to setup the gdb build.

pfalcon, 2015-10-30:
No idea either. But if you find out, feel free to submit README patch - that's useful.

jcmvbkbc, 2015-10-30:
Go to crosstool-NG directory and run ct-ng menuconfig, find CT_GDB_CROSS_PYTHON entry and turn it on. I'll update crosstool configs.

rabarar, 2015-10-30:
Thanks Paul,

So I did that, saved it, and then I did a make clean, and  a  make STANDALONE=n

Sound right?

> On Oct 30, 2015, at 1:46 PM, Max Filippov <notifications@github.com> wrote:
> 
> Go to crosstool-NG directory and run ct-ng menuconfig, find CT_GDB_CROSS_PYTHON entry and turn it on. I'll update crosstool configs.
> 
>

rabarar, 2015-10-30:
Also,

Is there a fix to cleanup the Xcode compile problem on mac that requires the patching as below?

gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-blocking.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-clast-to-gimple.c
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-dependences.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-interchange.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-optimize-isl.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-poly.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-scop-detection.c 
gsed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite-sese-to-poly.c

I’m using the most recent Xcode 
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.0.0 (clang-700.1.76)
Target: x86_64-apple-darwin15.0.0
Thread model: posix

The compile breaks, I run the patch that adds stddef.h to the files listed above, and off to the races from there to complete the build…  but i imagine that’s gotta be frustrating to folks who have no clue why the build fails…

Rob

pfalcon, 2015-10-30:
> So I did that, saved it, and then I did a make clean, and a make STANDALONE=n

Unlikely. @jcmvbkbc suggests to build it manually and verify that the result is expected. If it is, you can put that config option to crosstool-config-overrides file, and then it can be used in unattended mode (like, being automagically applied after "make clean" on esp-open-sdk side).

rabarar, 2015-10-30:
That did it! Thanks again for the quick reply!