pfalcon/esp-open-sdk #106: Cannot build when LD_LIBRARY_PATH is set

Phyks, 2015-11-11:
It is common to set `LD_LIBRARY_PATH`, however build errors if it is set. Why not backupping it, resetting it and restoring it after the build?

Thanks

pfalcon, 2015-11-11:
It's not common to set it. If you set it, dealing with it is your problem.

closed 2015-11-11 by pfalcon

Phyks, 2015-11-11:
Well, it is common if you use some tools, e.g. Torch. Dealing with it is not much overhead on esp-open-sdk side and I think it could be worth it.

pfalcon, 2015-11-11:
Torch? What Torch? Let them know that they do it wrong. In the meantime, build on clean, Torch-less system, only then mix in that Torch, if you have to.

> Dealing with it is not much overhead on esp-open-sdk side and I think it could be worth it.

Next report: "esp-open-sdk requires computer to build. Why not make it build without computer? It should be easy."

Phyks, 2015-11-11:
Link added in previous comment, it is a Lua library.

Anyway, guess you made your mind and it is useless to discuss it further.