pfalcon/esp-open-sdk #107: Tool Chain build fails - MPFR cannot be reached

tiestvangool, 2015-11-11:
I am unable to rebuild the ESP8266 tool chain. I created as outlined and build the tool chain successfully the first time. Had some issues with the VM, rebuild the VM and ran the same script and I am no longer able to build the tool chain. I have tried to build the tool chain on multiple machines and both the expressif sdk as well as the esp-open-sdk give the very same error, see details in the build log below.

When looking in the build log I can see that the mpfr.org site is being called which seems down. Anyone any suggestions how to resolve this?

....
[EXTRA]  Dumping internal crosstool-NG configuration: done in 0.09s (at 00:02)
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[DEBUG]    Already have 'gmp-5.1.3'
[EXTRA]    Retrieving 'mpfr-3.1.2'
[DEBUG]    Trying 'http://www.mpfr.org/mpfr-current//mpfr-3.1.2.tar.xz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/Espressif/crosstool-NG/.build/tarballs/mpfr-3.1.2.tar.xz.tmp-dl' 'http://www.mpfr.org/mpfr-current//mpfr-3.1.2.tar.xz' 
[ALL  ]    --2015-11-11 14:50:59--  http://www.mpfr.org/mpfr-current//mpfr-3.1.2.tar.xz
[ALL  ]    Resolving www.mpfr.org (www.mpfr.org)... 152.81.144.6
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2015-11-11 14:51:10--  (try: 2)  http://www.mpfr.org/mpfr-current//mpfr-3.1.2.tar.xz
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... failed: Connection timed out.
[ALL  ]    Resolving www.mpfr.org (www.mpfr.org)... 152.81.144.6
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... failed: Connection timed out.
[ALL  ]    Retrying.
[ALL  ]    
[ALL  ]    --2015-11-11 14:51:32--  (try: 3)  http://www.mpfr.org/mpfr-current//mpfr-3.1.2.tar.xz
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... failed: Connection timed out.
[ALL  ]    Giving up.
[ALL  ]    
...

[ALL  ]    --2015-11-11 15:01:53--  (try: 3)  http://www.mpfr.org/mpfr-3.1.2//mpfr-3.1.2
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... failed: Connection timed out.
[ALL  ]    Resolving www.mpfr.org (www.mpfr.org)... 152.81.144.6
[ALL  ]    Connecting to www.mpfr.org (www.mpfr.org)|152.81.144.6|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
[ALL  ]    Giving up.
[ALL  ]    
[DEBUG]    Not at this location: "http://www.mpfr.org/mpfr-3.1.2//mpfr-3.1.2"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_mpfr_get[scripts/build/companion_libs/110-mpfr.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@583]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 11:16.10)

pfalcon, 2015-11-11:
https://github.com/pfalcon/esp-open-sdk/issues/100

closed 2015-11-11 by pfalcon

tiestvangool, 2015-11-11:
Paul,

I understand this might not be related to the ESP-OPEN-SDK but could you guide/coach me into a direction on how to resolve this?

Thanks a ton,
-Tiest

pfalcon, 2015-11-11:
How about doing a search on your side (that's what *any* guide on reporting bugs suggests as 1st step)? https://github.com/pfalcon/esp-open-sdk/issues/45#issuecomment-155919502

tiestvangool, 2015-11-12:
Thank you - I am pretty new in this hence my questioning! Thanks for the help - next up trying to change the shell script to pull from different location.... Again thanks!
-Tiest

Namphibian, 2015-11-14:
@tiestvangool  @pfalcon  I had a exactly the same issue two days ago. I then tried to access the mpfr url in the build scripts and it appeared that mpfr website was affected by some networking issues appeared to be DNS related but I was unable to confirm. I tried it a few hours later and all worked.

nitrotm, 2015-11-14:
Same issue here right now. Isn't there a mirror archive for mpfr?

billfarrow, 2015-11-15:
The problem is in the crosstool-NG project and not esp-open-sdk project. You can replace the www.mpfr.org urls in crosstool-ng scripts with the GNU mirror: http://ftp.gnu.org/gnu/mpfr/

Edit these files and run make again.
crosstool-NG/scripts/build/companion_libs/110-mpfr.sh
crosstool-NG/lib/ct-ng.1.20.0/scripts/build/companion_libs/110-mpfr.sh