pfalcon/esp-open-sdk #109: Support for non OS SDK 1.5

nkolban, 2015-11-28:
With the release of the SDK 1.5 from Espressif ... see

http://bbs.espressif.com/viewtopic.php?f=46&t=1442

Does anything special have to be done to leverage that SDK?  Does this project have a flag or mechanism to declare which SDK should be used when building the environment?

pfalcon, 2015-11-28:
I thought the book author is who knows it all! ;-)

> Does anything special have to be done to leverage that SDK?

Nothing "special", only the usual stuff - try it, make sure that patches apply, test that application(s) built against it work, ship it. Most recent vendor SDK updates were handled by the community, so I'll leave that to it (especially that I don't have access to esp8266 now).

> Does this project have a flag or mechanism to declare which SDK should be used when building the environment?

Yes: https://github.com/pfalcon/esp-open-sdk/blob/master/Makefile#L3

nkolban, 2015-11-28:
Awesome ... thanks for the response ... that looks perfect.  I'm much more a librarian than a skilled artificer.

closed 2015-11-28 by nkolban