pfalcon/esp-open-sdk #114: sed error on mac os x option -r -i....

MaBecker, 2015-12-18:
replace sed with gsed

Daubsy, 2015-12-18:
Can you elaborate on how to fix this? Im stuck.

MaBecker, 2015-12-18:
vi Makefile
 
       search for sed and replace with gsed

	@gsed -e 's/\r//' sdk/ld/eagle.app.v6.ld | gsed -e s@../ld/@@ >$(TOOLCHAIN)/xtensa-lx106-elf/sysroot/usr/lib/eagle.app.v6.ld
	@gsed -e 's/\r//' sdk/ld/eagle.rom.addr.v6.ld >$(TOOLCHAIN)/xtensa-lx106-elf/sysroot/usr/lib/eagle.rom.addr.v6.ld
	gsed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="$(TOOLCHAIN)"% .config
	gsed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config

Daubsy, 2015-12-18:
I tried that but im still getting the error output below. Not sure what it means but it seems related to sed.

```
make -C crosstool-NG -f ../Makefile _ct-ng
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-apple-darwin14.5.0
checking host system type... x86_64-apple-darwin14.5.0
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for a sed that does not truncate output... /usr/bin/sed
checking whether sed understands -r -i -e... no
configure: error: 
make[1]: *** [_ct-ng] Error 1
make: *** [crosstool-NG/ct-ng] Error 2
```

MaBecker, 2015-12-18:
I used: make STANDALONE=n

vnitinv, 2016-01-02:
I am getting below error while doing make on Mac

[INFO ]  Installing pass-2 core C compiler
[ERROR]    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstddef:46:9: error: no member named 'ptrdiff_t' in the global namespace
[ERROR]    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstddef:46:9: error: no member named 'ptrdiff_t' in the global namespace
[ERROR]    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator:413:13: error: unknown type name 'ptrdiff_t'

jellyfishcoder, 2016-06-10:
Do `brew install gnu-sed --with-default-names`

closed 2019-06-18 by MaBecker