pfalcon/esp-open-sdk #115: build failure after clone: C compiler cannot create executables

susisstrolch, 2015-12-19:
Can't build anymore on Linux (Mint 17.2) after cloning:
```
...Repos/esp-open-sdk $ git submodule sync 
Synchronizing submodule url for 'crosstool-NG'
Synchronizing submodule url for 'esptool'
Synchronizing submodule url for 'lx106-hal'
...Repos/esp-open-sdk $ git submodule update
Submodule path 'esptool': checked out '4334f05a91d60718217ffce95a59329792ea8893'
...Repos/esp-open-sdk $ make STANDALONE=y
```
The crosstool-NG submodule show the proper version:
```
Repos/esp-open-sdk/crosstool-NG $ git branch
* (detached from feb1fb8)
  xtensa-1.22.x
```
And here's the build output:
```
[INFO ]  Build started 20151219.135259
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 0.03s (at 00:00)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 22.88s (at 00:23)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 19.98s (at 00:43)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 9.40s (at 00:53)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 13.19s (at 01:06)
[INFO ]  =================================================================
[INFO ]  Installing CLooG for host
[INFO ]  Installing CLooG for host: done in 4.56s (at 01:11)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 3.79s (at 01:14)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 29.38s (at 01:44)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C compiler
[ERROR]    configure: error: in `/home/strolch/Development/Espressif/share/Repos/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-core-pass-2':
[ERROR]    configure: error: C compiler cannot create executables
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@387]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 1:44.31)
[01:44] / make[2]: *** [build] Error 77
make[2]: Leaving directory `/home/strolch/Development/Espressif/share/Repos/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/home/strolch/Development/Espressif/share/Repos/esp-open-sdk/crosstool-NG'
make: *** [/home/strolch/Development/Espressif/share/Repos/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```

susisstrolch, 2015-12-22:
solved - the proper libstdc++ for host compiler was missing...

closed 2015-12-22 by susisstrolch

lnsriravichandran22, 2016-03-10:
So , How this Issue can be fixed as I encounter the same!!