pfalcon/esp-open-sdk #122: OS X - overloading error in pass-2 core C compiler

BroukPytlik, 2016-01-11:
I used the [OSX howto](http://www.esp8266.com/wiki/doku.php?id=setup-osx-compiler-esp8266) steps (the Brew variant). But when I try to compile, no matter whether with STANDALONE=y or n, I get the following errors. Also, I tried it both with and without the patches described in MacPorts section of the link in case the page is incorrectly formatted and the patches belongs to both variants, but it changed nothing.

I suppose I may have something wrong on my side, so I will be thankful for any hint where to look next.

OS X El Capitan, using Brew for dependencies.

```
[INFO ]  Installing pass-2 core C compiler
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/system.h:500:20: error: functions that differ only in their return type cannot be overloaded
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/statistics.h:25:2: error: GATHER_STATISTICS must be defined
[ERROR]    make[3]: *** [ggc-none.o] Error 1
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/system.h:500:20: error: functions that differ only in their return type cannot be overloaded
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/system.h:500:20: error: functions that differ only in their return type cannot be overloaded
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/system.h:500:20: error: functions that differ only in their return type cannot be overloaded
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/system.h:500:20: error: functions that differ only in their return type cannot be overloaded
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include/line-map.h:207:38: error: use of undeclared identifier 'CHAR_BIT'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include/line-map.h:207:38: error: use of undeclared identifier 'CHAR_BIT'
[ERROR]    make[3]: *** [prefix.o] Error 1
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include/line-map.h:207:38: error: use of undeclared identifier 'CHAR_BIT'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/../libcpp/include/line-map.h:207:38: error: use of undeclared identifier 'CHAR_BIT'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/statistics.h:25:2: error: GATHER_STATISTICS must be defined
[ERROR]    make[3]: *** [xtensa-common.o] Error 1
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:295:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/statistics.h:25:2: error: GATHER_STATISTICS must be defined
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:303:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:317:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    make[3]: *** [c/gccspec.o] Error 1
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:295:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:303:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/vec.h:317:7: error: use of undeclared identifier 'GATHER_STATISTICS'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:1713:10: error: use of undeclared identifier 'open'; did you mean 'zopen'?
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:1713:8: error: assigning to 'int' from incompatible type 'FILE *' (aka '__sFILE *')
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:5058:19: error: use of undeclared identifier 'HAVE_GNU_LD'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:8128:10: error: use of undeclared identifier 'strverscmp'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:8294:8: error: use of undeclared identifier 'open'
[ERROR]    /Volumes/eep-open-sdk/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.2/gcc/gcc.c:8334:27: error: use of undeclared identifier 'CHAR_BIT'
[ERROR]    make[3]: *** [gcc.o] Error 1
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: do_cc_core_backend[scripts/build/cc/gcc.sh@464]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc/gcc.sh@153]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@638]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 12:05.00)
[12:05] / make[2]: *** [build] Error 1
make[1]: *** [_toolchain] Error 2
make: *** [/Volumes/eep-open-sdk/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```

roosen, 2016-02-10:
Probably duplicate of #45

myfreeweb, 2017-08-27:
`GATHER_STATISTICS` (this) doesn't seem to have anything to do with `ptrdiff_t` / `__need_size_t` (#45).

I'm having this exact problem on FreeBSD 11 right now.