pfalcon/esp-open-sdk #123: Outdated crosstool-NG

geissdoerfer, 2016-01-18:
The commit you're using for crosstool-NG is outdated and apparently one of the companion_lib (mpc) download scripts points to a website, that is no longer maintained (www.multiprecision.org). The build process fails.

See:
https://github.com/jcmvbkbc/crosstool-NG/blob/feb1fb829d5cb4f8739e488cbd2f0b72304705bb/scripts/build/companion_libs/140-mpc.sh

This was fixed in jcmvbkbc/crosstool-NG@736528197319aec938f459da04bbc438781d78bf

So you might want to use a newer version for this.

geissdoerfer, 2016-01-19:
I just checked and noticed the website is up again! However, this seems to happen from time to time, so i'll leave this open as you still might want to update

BroukPytlik, 2016-01-19:
I second this. In fact, I was unable to get the old version to compile - I had to switch to the current crosstool-NG to get this SDK working. I already created a pull request #126.

larsenglund, 2016-02-18:
I tried replacing the crosstool-NG folder with the one from https://github.com/jcmvbkbc/crosstool-NG but I get this:
```
[INFO ]  Installing pass-2 core C gcc compiler
[ERROR]    make[4]: *** [libbackend.a] Error 1
[ERROR]    make[3]: *** [all-gcc] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@542]
[ERROR]  >>        called from: do_gcc_core_pass_2[scripts/build/cc/100-gcc.sh@160]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc.sh@42]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-46-g216e2fc/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 16:32.86)
[16:33] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/lars/esp-open-sdk/crosstool-NG'
../Makefile:111: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/lars/esp-open-sdk/crosstool-NG'
Makefile:108: recipe for target '/home/lars/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/lars/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```
Any suggestions?

jrobeson, 2016-10-19:
crosstool-ng has certainly been updated since then, but the current reference is now 5 months old.  Should it be bumped, or is there anything actionable on this ticket otherwise?

pfalcon, 2016-10-19:
Let's close this.

closed 2016-10-19 by pfalcon