pfalcon/esp-open-sdk #132: MacOS - ar error unpacking libcrypto.a

MitchBradley, 2016-02-09:
I got the following error when running make on MacOS 10.11.3 with XCode 7.2.1

   cd esp_iot_sdk_v1.5.0/lib; mkdir -p tmp; cd tmp; ar x ../libcrypto.a; cd ..; ar rs libwpa.a tmp/*.o
   ar: /: Is a directory
   ar: //: Is a directory
   ar: aes-cbc.o/: No such file or directory
   ... many more similar errors

It appears that /usr/bin/ar on my system does not handle the particular archive format very well.

I worked around it by replacing "ar" with "gar" in the Makefile rule for .sdk_patch_1.5.0 , i.e.

   cd $(VENDOR_SDK_DIR)/lib; mkdir -p tmp; cd tmp; gar x ../libcrypto.a; cd ..; gar rs libwpa.a tmp/*.o

roosen, 2016-02-10:
I would suggest to change ar by $(TOOLCHAIN)/bin/xtensa-lx106-elf-ar. Should work on all platforms.

pfalcon, 2016-02-10:
Good suggestion, I will look into it.

pfalcon, 2016-02-11:
Fixed.

closed 2016-02-11 by pfalcon