pfalcon/esp-open-sdk #136: "Corrupt" esp_sdk_ver.h

ieei, 2016-02-18:
The SDK is patched by one of the make targets `.sdk_patch_*` which uses echo with the -e option.
My system (or my /bin/sh) doesn't support -e. I don't think -e is defined by **POSIX** either. This means that `esp_sdk_ver.h` looks like this:
```
-e #undef ESP_SDK_VERSION
#define ESP_SDK_VERSION 010502
```

I guess forcing make to use bash instead of /bin/sh or replacing /bin/sh with bash or another shell would mitigate the issue for now.

pfalcon, 2016-02-19:
Well, if you you've done a bit of system usage with shell set not to /bin/bash, you already saw a bunch of issues like that, so this one should not surprise you. README pretty clearly says that GNU/POSIX system is required, not just POSIX. Feel free to provide patches though (please make sure they don't increase number of lines in O(N) manner, where N is number of SDK versions; refactor first if needed).

pfalcon, 2016-03-13:
In the meantime, made explicit a requirement of bash.

gschmottlach, 2016-03-13:
This bit me too. If you want your Makefile's to use the bash shell to execute its shell operations then you must explicitly set the "SHELL" variable in your Makefile, e.g.

~~~~
SHELL=/bin/bash
~~~~

Without this, `make` will use /bin/sh which does **not** define the '-e' option for echo. As a result, we see

~~~~
-e #undef ESP_SDK_VERSION
~~~~

littered in the  esp_sdk_ver.h file.  See this [link](https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html) for more details on choosing your shell.

dpgeorge, 2016-03-13:
One solution is to not use "-e" at all and split the echo into two commands, one for each line to print.  Factoring the echo to be part of the "sdk_patch" target would be a good solution to reduce code size.  Might also be able to automatically deduce the version id from the "vendor_sdk" variable.

Tkizzy, 2016-03-14:
Just piling on. I have the same error.

```
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:1: error: expected identifier or '(' before '-' token
 -e #undef ESP_SDK_VERSION
 ^
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:4: error: stray '#' in program
 -e #undef ESP_SDK_VERSION

```

Tkizzy, 2016-03-14:
the suggested fix to make `SHELL =/bin/bash` in the makefile...could you elaborate?

does it matter where it's put in the file micropython/esp8266/Makefile ? at the top? at the end? Putting it at the top doesn't seem to fix it for me.

EklundJL, 2016-03-27:
@Tkizzy  - Were you able to eventually build micropython?

jurviz, 2016-04-15:
I'm hit by the same error. Trying to set /bin/bash as shell as suggested does not help.
Can somebody explain what to do?

jurviz, 2016-04-15:
After updating to latest state as described in README, the error did not reoccur and the build finished without errors.

rigid, 2016-05-26:
so this can be closed?

EklundJL, 2016-05-26:
Yes

ieei, 2016-05-26:
Fixed as of a890758ed80c9657fdc5d0235edc64ba5ec1bd7b as far as I'm concerned.

closed 2016-05-26 by ieei

pfalcon, 2016-05-26:
Note that I'm personally not too sure that this issue is "fixed". What fixed it for macosx, broke it for Fedora: https://github.com/pfalcon/esp-open-sdk/issues/152 . So, if Fedora complaints continue, a890758 may be as well reverted.