pfalcon/esp-open-sdk #137: make error

lishidetiankong, 2016-03-03:
Hi:    
     I git clone this repositories and execute the command as your mension , but there is a error happen,
     the error shown just like below!

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory `/home/yang/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/yang/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory `/home/yang/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[ERROR]  Don't set LD_LIBRARY_PATH. It screws up the build.
[00:00] / touch: cannot touch '/home/yang/esp-open-sdk/crosstool-NG/.build/backtrace': No such file or directory
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
[ERROR]  >>        called from: CT_TestAndAbort[scripts/functions@351]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@55]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 24283229:01.28)
[00:00] / make[2]: *** [build] Error 1
make[2]: Leaving directory `/home/yang/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/home/yang/esp-open-sdk/crosstool-NG'
make: *** [/home/yang/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
yang@yang-B85M-D2V:~/esp-open-sdk$

pfalcon, 2016-03-03:
Please read the error message and comply.

closed 2016-03-03 by pfalcon

PeterQFR, 2017-02-16:
What pfalcon means is that you have specified LD_LIBRARY_PATH in you environment. to check type 
$ echo $LD_LIBRARY_PATH

if anything is returned, then the variable has been set, usually in .bashrc. Change this in .bashrc and you'll probably need to open a new window to eliminated the variable.

MrDadaGuy, 2017-09-19:
In case anyone else finds this and still looking for a solution:

I had this problem also - I ran "unset LD_LIBRARY_PATH" then had to delete esp-open-sdk directory, then had to re-run the command.  Good luck!

Chrisys93, 2018-01-29:
I'm still having problems with this, even though it does not mention anything about my env variables, just "Unknown stat format options", as follows:

chrisys1993@Chris:~/pi$ ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180129.153142
[INFO ]  Building environment variables
[EXTRA]  Preparing working directories
[ERROR]  Unknown stat format options
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@484]
[ERROR]  >>        called from: CT_DoForceRmdir[scripts/functions@617]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@281]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      https://crosstool-ng.github.io/docs/known-issues/
[ERROR]  >>
[ERROR]  >>  If you feel this is a bug in crosstool-NG, report it at:
[ERROR]  >>      https://github.com/crosstool-ng/crosstool-ng/issues/
[ERROR]  >>
[ERROR]  >>  Make sure your report includes all the information pertinent to this issue.
[ERROR]  >>  Read the bug reporting guidelines here:
[ERROR]  >>      http://crosstool-ng.github.io/support/
[ERROR]
[ERROR]  (elapsed: 0:00.73)
[00:04] / /home/chrisys1993/.local/bin/ct-ng:142: recipe for target 'build' failed
make: *** [build] Error 1
chrisys1993@Chris:~/pi$

I am very confused by this point. I am trying to obtain a better toolchain config. for cross-compiling for my raspberry pi 3 B. I have followed these instructions almost to the letter (apart from versions where I couldn't actually go lower or was inappropriate to do so): https://elinux.org/RPi_Linaro_GCC_Compilation
up to the build command, of course.
Hope someone can help here.
Thank you in advance.

Regards,
Chris

BorisHelloworld, 2018-04-02:
It tells you to check the build.log, the will be an extra line of [DEBUG] telling you what is clashing with the build. For me, I specified LD_LIBRARY_PATH in ~/.bashrc for STLinkUtil before, so I simply commented out that line. Then go back to make esp sdk and it worked.

akhilrb, 2018-11-25:
In my case, I had ROS Melodic installed before compiling the xtensa toolchain.
The official ROS installation pages recommend appending the following line into ~/.bashrc :
`source /opt/ros/melodic/setup.bash`
so that the ROS environment variables are automatically added to a bash session every time a new shell is launched.
For a beginner, it can be difficult realising that even though there's no variable LD_LIBRARY_PATH actually mentioned in the ~/.bashrc, it is being sourced from ROS's (or any other install's) files.
Commenting those specific lines fixes the problem.