pfalcon/esp-open-sdk #138: Cannot build latest version (c_types.h missing)

nitrotm, 2016-03-05:
make[4]: Leaving directory '/tmp/esp-open-sdk/src/git/lx106-hal'
make[3]: Leaving directory '/tmp/esp-open-sdk/src/git/lx106-hal'
make[2]: Leaving directory '/tmp/esp-open-sdk/src/git/lx106-hal'
make[1]: Leaving directory '/tmp/esp-open-sdk/src/git/lx106-hal'
make -C esp-open-lwip -f Makefile.open install \
    CC=/tmp/esp-open-sdk/src/git/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc \
    PREFIX=/tmp/esp-open-sdk/src/git/xtensa-lx106-elf
make[1]: Entering directory '/tmp/esp-open-sdk/src/git/esp-open-lwip'
/tmp/esp-open-sdk/src/git/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals   -c -o lwip/core/def.o lwip/core/def.c
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
                     ^
compilation terminated.
<builtin>: recipe for target 'lwip/core/def.o' failed
make[1]: Leaving directory '/tmp/esp-open-sdk/src/git/esp-open-lwip'
make[1]: *** [lwip/core/def.o] Error 1
Makefile:283: recipe for target 'lwip' failed
make: *** [lwip] Error 2

drasko, 2016-03-05:
Same here:
```
make[1]: Leaving directory '/home/drasko/esp8266/esp-open-sdk/lx106-hal'
make -C esp-open-lwip -f Makefile.open install \
    CC=/home/drasko/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc \
    PREFIX=/home/drasko/esp8266/esp-open-sdk/xtensa-lx106-elf
make[1]: Entering directory '/home/drasko/esp8266/esp-open-sdk/esp-open-lwip'
/home/drasko/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals   -c -o lwip/core/def.o lwip/core/def.c
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
                     ^
compilation terminated.
<builtin>: recipe for target 'lwip/core/def.o' failed
make[1]: *** [lwip/core/def.o] Error 1
make[1]: Leaving directory '/home/drasko/esp8266/esp-open-sdk/esp-open-lwip'
Makefile:283: recipe for target 'lwip' failed
make: *** [lwip] Error 2
drasko@Lenin:~/esp8266/esp-open-sdk$ 
```

drasko, 2016-03-05:
A quick and dirty fix is to patch `Makefile.open` in the `esp-open-lwip` dir:
```
drasko@Lenin:~/esp8266/esp-open-sdk/esp-open-lwip$ git diff
diff --git a/Makefile.open b/Makefile.open
index d398220..d21eeb0 100644
--- a/Makefile.open
+++ b/Makefile.open
@@ -1,7 +1,7 @@
 CC = xtensa-lx106-elf-gcc
 DEFS = -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH
 COPT = -Os
-CFLAGS = $(DEFS) $(COPT) -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals
+CFLAGS = $(DEFS) $(COPT) -Iinclude -I../esp_iot_sdk_v1.5.2/include -Wl,-EL -mlongcalls -mtext-section-literals
 # Install prefix of esp-open-sdk toolchain
 PREFIX = ~/toolchain/xtensa-lx106-elf
 
drasko@Lenin:~/esp8266/esp-open-sdk/esp-open-lwip$ 
```

I'll let @pfalcon  fix this as here probably some env variable of `VENDOR_SDK_DIR` has to be propagated.

mvphatak, 2016-03-05:
After adding this patch I am getting following error: -- I may not have added patch correctly.
cp esptool/esptool.py /esptools/esp-open-sdk/xtensa-lx106-elf/bin/
ln -snf esp_iot_sdk_v1.5.2 sdk
make -C esp-open-lwip -f Makefile.open install \
	    CC=/esptools/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc \
	    PREFIX=/esptools/esp-open-sdk/xtensa-lx106-elf
/esptools/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc    -c -o lwip/core/def.o lwip/core/def.c
lwip/core/def.c:39:22: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
                      ^
compilation terminated.
make[1]: *** [lwip/core/def.o] Error 1
make: *** [lwip] Error 2

PTDreamer, 2016-03-05:
Same problem here but only if compiled with "STANDALONE=n"
@drasko patch fixes the issue.

pfalcon, 2016-03-06:
A change which should fix this was pushed. If you still have any
issues, be sure to provide all the steps required to reproduced the
issue on a fresh system.

mvphatak, 2016-03-06:
It works with "STANDALONE=y" but same issue otherwise.

Ghostbird, 2016-03-06:
It's broken for me too, on a fresh clone. Note that I built it with ```STANDALONE=y```

mvphatak, 2016-03-06:
@Ghostbird  You still have to use solution from https://github.com/pfalcon/esp-open-sdk/issues/94 to fix the issue for Installing pass-2 core C compiler.. Once you do that it works.

Ghostbird, 2016-03-06:
Ah, I had to use sed instead of gsed, and I just used a rather sloppy one-liner, but it worked.
```bash
sed -i '1s/^/#include <stddef.h>\n/' crosstool-NG/.build/src/gcc-4.8.2/gcc/graphite*.c
```

deece, 2016-03-31:
This still occurs on the latest master.

Please see the attached console.log to confirm the following:
- Fresh checkout of master at:
           7235b03 esptool: Update to upstream (ebf02c92).

I've also included the full build log, the important part is:
make -C esp-open-lwip -f Makefile.open install \
    CC=/home/deece/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc \
    PREFIX=/home/deece/esp8266/esp-open-sdk/xtensa-lx106-elf
make[1]: Entering directory '/home/deece/esp8266/esp-open-sdk/esp-open-lwip'
/home/deece/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -DLWIP_OPEN_SRC -DPBUF_RSV_FOR_WLAN -DEBUF_LWIP -DICACHE_FLASH -Os -Iinclude -Wl,-EL -mlongcalls -mtext-section-literals   -c -o lwip/core/def.o lwip/core/def.c
In file included from include/lwip/arch.h:43:0,
                 from include/lwip/debug.h:35,
                 from include/lwip/opt.h:46,
                 from lwip/core/def.c:39:
include/arch/cc.h:38:21: fatal error: c_types.h: No such file or directory
 #include "c_types.h"
                     ^
compilation terminated.
make[1]: *** [lwip/core/def.o] Error 1
<builtin>: recipe for target 'lwip/core/def.o' failed
make: *** [lwip] Error 2
make[1]: Leaving directory '/home/deece/esp8266/esp-open-sdk/esp-open-lwip'
Makefile:286: recipe for target 'lwip' failed

[logs.zip](https://github.com/pfalcon/esp-open-sdk/files/197502/logs.zip)

nitrotm, 2016-04-02:
Pull request #148 should fix this issue.

sappo, 2016-04-05:
This issue is really annoying. Can anybody tell which is the last commit this did work, then I'll revert to that.

pfalcon, 2016-04-05:
@sappo: You should just build with "make" and there're no issues.

sappo, 2016-04-05:
@pfalcon Thanks for the quick response. Just "make" works fine indeed :+1:

pfalcon, 2016-04-22:
This was resolved.

closed 2016-04-22 by pfalcon