pfalcon/esp-open-sdk #139: build failed in step 'retreiving needed tool chain components' tarballs'

sfa1906, 2016-03-07:
Hey guys the build step is failing. I tried multiple times but it is not working. It is failing exactly 5min and 27secs after starting each time.. please suggest some solution.. I have checked the internet connection and it is perfectly fine. Thanks in advance..

`[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_gmp_get[scripts/build/companion_libs/100-gmp.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@583]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 5:25.77)
[05:27] / make[2]: *** [build] Error 1
make[2]: Leaving directory `/home/fareed/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/home/fareed/esp-open-sdk/crosstool-NG'
make: *** [/home/fareed/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
`

jeffryr, 2016-03-10:
Transient upstream problem?  I just did a clean git clone followed by make STANDALONE=y on Linux Mint 17.3 x64

```
git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
time -p make STANDALONE=Y
...
Xtensa toolchain is built, to use it:

export PATH=/home/jra/ESP-8266/esp-open-sdk-standalone/xtensa-lx106-elf/bin:$PATH

Espressif ESP8266 SDK is installed, its libraries and headers are merged with the toolchain

real 1080.06
user 1668.08
sys 191.45
```

closed 2016-03-30 by sfa1906

HIFZA-SHAIKH, 2016-08-23:
Thanks .....had a successful built