pfalcon/esp-open-sdk #141: Issue with issuing make command

lnsriravichandran22, 2016-03-09:
Can Somebody please tell , in which location should I actually issue the "make" command??(I mean the first ever make command which is to be issued at the first step of building)

pfalcon, 2016-03-09:
It should be run in top-level project dir, like:

~~~
git clone https://github.com/pfalcon/esp-open-sdk
cd esp-open-sdk
make
~~~

That the (more or less obvious) default, if you don't see any specific instructions, that's how it is, for almost any project.

closed 2016-03-09 by pfalcon

lnsriravichandran22, 2016-03-09:
Thanks a lot pfalcon. But after issuing the "make" command, I am receiving some error and it is related to tool chain stuff it seems. If you don't mind, I could use your hand-holding.

Please refer the screen-shot which I have attached.
Thanks in Advance!!

[image]

lnsriravichandran22, 2016-03-10:
I still can't get out of this issue and I can't understand why is this happening as I am a little bit new to these own sdk concepts. Can someone please help me with a permanent solution for this??

Thanks in Advance!!