pfalcon/esp-open-sdk #142: ERROR: Empty preprocessor output - check for errors above

Tkizzy, 2016-03-13:
I've successfully generated the toolset on OSX, and set 

`echo "PATH=$(pwd)/xtensa-lx106-elf/bin:\$PATH" >> ~/.bash_profile`

I've done `cd micro python/esp8266` but when I try to do `make` I get the errors below.

```
make: xtensa-lx106-elf-gcc: Command not found
GEN build/genhdr/qstrdefs.generated.h
/bin/sh: xtensa-lx106-elf-gcc: command not found
ERROR: Empty preprocessor output - check for errors above
make: *** [build/genhdr/qstrdefs.generated.h] Error 1
make: *** Deleting file `build/genhdr/qstrdefs.generated.h'
```

Help me build this firmware!

dpgeorge, 2016-03-13:
Did you restart your shell session to make the bash_profile settings take effect?  You can check by doing: echo $PATH.

Tkizzy, 2016-03-13:
@dpgeorge Thanks for your response. I sure did. 'echo $PATH' returns ``` /Users/me/Documents/gitStuff/pySerialStuff/micropython/esp8266/xtensa-lx106-elf/bin:/Users/me/Documents/gitStuff/pySerialStuff/micropython/esp8266/xtensa-lx106-elf/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin```

dpgeorge, 2016-03-13:
Then if you type `xtensa-lx106-elf-gcc` at the command line it should work, does it?

Tkizzy, 2016-03-13:
It doesn't :( 

```
anty:~/Documents/gitStuff/pySerialStuff/micropython/esp8266 🚦  xtensa-lx106-elf-gcc

-bash: xtensa-lx106-elf-gcc: command not found

```

A file search doesn't yield any files by the name `xtensa-lx106-elf-gcc` should that be somewhere in the esp-open-sdk?

Tkizzy, 2016-03-13:
ok I see xtensa-lx106-elf-gcc in the esp-open-sdk folder where I built it in a case-sensitive disk image. Is there something pointing to somewhere there I need to add to my Path?

Tkizzy, 2016-03-14:
ok - fixed this by remaking the toolchain and then doing
`export PATH=/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/bin:$PATH`

H

Tkizzy, 2016-03-14:
OK. the make command works now but there's another error. 

```
LINK build/firmware.elf
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_print_meminfo':
(.irom0.text+0xfb0): undefined reference to `dhcps_start'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_get_free_heap_size':
(.irom0.text+0xfe4): undefined reference to `dhcps_start'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_get_chip_id':
(.irom0.text+0x1000): undefined reference to `dhcps_stop'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_get_rtc_time':
(.irom0.text+0x102c): undefined reference to `dhcps_stop'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_get_rtc_time':
(.irom0.text+0x1050): undefined reference to `dhcp_start'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x1098): undefined reference to `dhcp_start'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x10ac): undefined reference to `dhcp_stop'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x10d8): undefined reference to `dhcp_stop'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x10f4): undefined reference to `dhcp_cleanup'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x1118): undefined reference to `dhcp_stop'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `system_mktime':
(.irom0.text+0x1120): undefined reference to `dhcp_cleanup'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_dhcps_start':
(.irom0.text+0x122c): undefined reference to `netif_set_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_dhcps_stop':
(.irom0.text+0x1285): undefined reference to `netif_set_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_dhcps_stop':
(.irom0.text+0x1298): undefined reference to `netif_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_station_dhcpc_start':
(.irom0.text+0x131a): undefined reference to `netif_set_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_station_dhcpc_stop':
(.irom0.text+0x134d): undefined reference to `netif_set_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_set_default_ssid':
(.irom0.text+0x29d0): undefined reference to `ethbroadcast'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_get_config_default':
(.irom0.text+0x2d45): undefined reference to `netif_set_default'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_get_station_info':
(.irom0.text+0x2fc4): undefined reference to `netif_set_addr'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(user_interface.o): In function `wifi_softap_get_station_info':
(.irom0.text+0x3018): undefined reference to `netif_set_addr'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x50c): undefined reference to `sys_check_timeouts'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x540): undefined reference to `lwip_init'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x544): undefined reference to `espconn_init'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x5ce): undefined reference to `lwip_init'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x5d4): undefined reference to `espconn_init'
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libmain.a(app_main.o): In function `user_rf_pre_init':
[... cut ...]
modlwip.c:(.text.lwip_getaddrinfo+0x36): undefined reference to `dns_gethostbyname'
build/extmod/modlwip.o:(.text.lwip_socket_listen+0x4): undefined reference to `tcp_listen_with_backlog'
build/extmod/modlwip.o:(.text.lwip_socket_listen+0x8): undefined reference to `tcp_accept'
build/extmod/modlwip.o: In function `lwip_socket_listen':
modlwip.c:(.text.lwip_socket_listen+0x4d): undefined reference to `tcp_listen_with_backlog'
modlwip.c:(.text.lwip_socket_listen+0x68): undefined reference to `tcp_accept'
make: *** [build/firmware.elf] Error 1

```

Tkizzy, 2016-03-14:
the solution for the above was to do 

`cd esp-open-sdk`
`git checkout lwip-headers`

before building the toolkit

```
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:1: error: expected identifier or '(' before '-' token
 -e #undef ESP_SDK_VERSION
 ^
/Volumes/esp-open-sdk/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/esp_sdk_ver.h:1:4: error: stray '#' in program
 -e #undef ESP_SDK_VERSION
```