pfalcon/esp-open-sdk #145: "Retrieving needed toolchain components' tarballs" taking forever

Logmytech, 2016-03-31:
I'm trying to compile the SDK and it's taking forver (I quit after 30 minutes).
```
user@user-desktop ~/esp-open-sdk $ make
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.2/
make -C crosstool-NG -f ../Makefile _toolchain
make[1]: Entering directory `/home/user/esp-open-sdk/crosstool-NG'
./ct-ng xtensa-lx106-elf
make[2]: Entering directory `/home/user/esp-open-sdk/crosstool-NG'
  CONF  config/config.in
#
# configuration saved
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory `/home/user/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/user/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory `/home/user/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20160331.180556
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[32:58] - 
```

ctrowat, 2016-04-01:
I was having this issue a day ago, checking the build.log file showed it was having trouble downloading files from http://www.mpfr.org/...
Checking the site showed it was not responding.  I tried again later, the site was back up and it got past this point.

pfalcon, 2016-04-01:
See https://github.com/pfalcon/esp-open-sdk/issues/100

closed 2016-04-01 by pfalcon

Logmytech, 2016-07-17:
It looks like it is normal for this to take long.

It did compile successfully after over 30 minutes

SuWeipeng, 2019-12-11:
> I was having this issue a day ago, checking the build.log file showed it was having trouble downloading files from http://www.mpfr.org/...
> Checking the site showed it was not responding. I tried again later, the site was back up and it got past this point.

yes it is, I download all tarballs and put into [google drive](https://drive.google.com/drive/folders/1fLITIQTCJv_D3tsKqCo6Gs-HEOwMnGh8?usp=sharing), download them manually, and put into `.build/tarballs`