pfalcon/esp-open-sdk #157: Build fails on Mac OSX

hansamann, 2016-05-01:
I've followed the build procedure as much as possible, built a case-sensitive file system, etc... but I get this:

make -C crosstool-NG -f ../Makefile _ct-ng
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-apple-darwin15.4.0
checking host system type... x86_64-apple-darwin15.4.0
checking for a BSD-compatible install... /usr/bin/install -c
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for a sed that does not truncate output... /usr/bin/sed
checking whether sed understands -r -i -e... no
configure: error: 
make[1]: *** [_ct-ng] Error 1
make: *** [crosstool-NG/ct-ng] Error 2

Any ideas?

eadf, 2016-05-21:
I managed to build esp-open-sdk on OSX (macports) with [these build instructions.](http://www.esp8266.com/wiki/doku.php?id=setup-osx-compiler-esp8266)

jellyfishcoder, 2016-06-10:
Install home-brew, with `/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"` then run `brew update`, `brew install gcc` (which takes A LONG TIME), then run `brew install coreutils`, which installs command line tools normally in linux. Now, run `brew install help2man`. If it says your missing any other packages, just try `brew install packagename`.

mcauser, 2016-10-26:
May be unrelated... I was having trouble getting `make clean` to run on OS X.
Was failing at step `make -C crosstool-NG clean`.
Had to cd into `esp-open-sdk/crosstool-NG` and run `./configure` to regenerate it's `Makefile` as it has hard coded paths to binaries.

Why? It was after a Boxen upgrade, which moved Homebrew to a different folder.
(`/opt/boxen/homebrew/` -> `/usr/local/Cellar`)
After that, `make clean` and `make` worked again.