pfalcon/esp-open-sdk #163: build on freebsd

sticilface, 2016-05-18:
I'm very new to this, build doesn't work on freebsd?  

does anyone have any experience of building on this? 

following error

```
root@test:~/esp-open-sdk # make STANDALONE=y
"Makefile", line 76: Missing dependency operator
"Makefile", line 82: Need an operator
"Makefile", line 85: Need an operator
"Makefile", line 88: Missing dependency operator
"Makefile", line 89: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 90: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 91: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 92: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 93: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 94: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 95: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 96: Need an operator
"Makefile", line 296: Missing dependency operator
"Makefile", line 300: warning: duplicate script for target "ifeq" ignored
"Makefile", line 300: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 303: warning: duplicate script for target "ifeq" ignored
"Makefile", line 303: warning: duplicate script for target "(y,y)" ignored
"Makefile", line 304: Need an operator
make: fatal errors encountered -- cannot continue
```

rigid, 2016-05-26:
maybe try to use gmake instead of make

sticilface, 2016-05-27:
Some help from igrr and he got it working!  shall close.

closed 2016-05-27 by sticilface

myfreeweb, 2016-07-30:
This shouldn't be closed… I figured out that this particular error is caused by the fact that even if I use `gmake`, the Makefile itself calls `make`. So the Makefile should be edited to pull out `make` into a variable & use `gmake` if it exists, otherwise use `make`!

Also `gsed` instead of `sed`, `gpatch` instead of `patch`.