pfalcon/esp-open-sdk #166: AT example Makefile doesn't compile

rigid, 2016-05-26:
I'm trying to compile the at_v0.20_on_SDKv0.9.3 example with the Makefile in examples/
It fails with:

```
$ make
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/at_baseCmd.o user/at_baseCmd.c
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/at_cmd.o user/at_cmd.c
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/at_ipCmd.o user/at_ipCmd.c
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/at_wifiCmd.o user/at_wifiCmd.c
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/at_port.o user/at_port.c
xtensa-lx106-elf-gcc -Os -Iinclude -mlongcalls   -c -o user/user_main.o user/user_main.c
xtensa-lx106-elf-gcc -Teagle.app.v6.ld driver/uart.o user/at_baseCmd.o user/at_cmd.o user/at_ipCmd.o user/at_wifiCmd.o user/at_port.o user/user_main.o -o at -nostdlib -Wl,--start-group -lmain -lupgrade -lnet80211 -lwpa -llwip -lpp -lphy -Wl,--end-group -lcirom -lgcc
/home/daniel/code/embedded/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: at section `.text' will not fit in region `iram1_0_seg'
/home/daniel/code/embedded/esp8266/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 6964 bytes
collect2: error: ld returned 1 exit status
Makefile:18: recipe for target 'at' failed
make: *** [at] Error 1
```

Is there a problem with the eagle.app.v6.ld script?

larsch, 2016-07-31:
The Makefile provided in the examples directory doesn't include the necessary compiler options to enable functions to be moved to the flash area of the binary. You'll need to add something like this to the Makefile:

```
CFLAGS += -mtext-section-literals \
  -finline-functions \
  -fdata-sections \
  -ffunction-sections \
  -DICACHE_FLASH
```

With these options, the RAM size goes from 00006fa8 down to 00006f30, and the flash section goes from 0002df40 up to 0002dfb0, with the blinky example.

I recommend also enabling optimization (for size):

```
CFLAGS += -Os
```

stevemk14ebr, 2016-08-10:
I'm having a similar issue with SDK 2.0.0, adding these flags has no effect on binary size. #189