pfalcon/esp-open-sdk #167: please update the README to explain which environment variables need to be unset

archenemies, 2016-05-27:
Looks like a promising tool!

This is as far as I got. I ran `make`, it said "Don't set LD_LIBRARY_PATH", so I ran it again this way:

    $ LD_LIBRARY_PATH= make
    cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
    make -C crosstool-NG -f ../Makefile _toolchain
    make[1]: Entering directory '/home/frederik/esp-open-sdk/crosstool-NG'
    ./ct-ng xtensa-lx106-elf
    make[2]: Entering directory '/home/frederik/esp-open-sdk/crosstool-NG'
      CONF  config/config.in
    #
    # configuration written to .config
    #

    ***********************************************************

    Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
    URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

    ***********************************************************

    WARNING! This sample may enable experimental features.
             Please be sure to review the configuration prior
             to building and using your toolchain!
    Now, you have been warned!

    ***********************************************************

    Now configured for "xtensa-lx106-elf"
    make[2]: Leaving directory '/home/frederik/esp-open-sdk/crosstool-NG'
    sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/frederik/esp-open-sdk/xtensa-lx106-elf"% .config
    sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
    cat ../crosstool-config-overrides >> .config
    ./ct-ng build
    make[2]: Entering directory '/home/frederik/esp-open-sdk/crosstool-NG'
    [INFO ]  Performing some trivial sanity checks
    [ERROR]  Don't set LIBRARY_PATH. It screws up the build.
    [ERROR]   
    [ERROR]  >>
    [ERROR]  >>  Build failed in step '(top-level)'
    [ERROR]  >>
    [ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
    [ERROR]  >>        called from: CT_TestAndAbort[scripts/functions@351]
    [ERROR]  >>        called from: main[scripts/crosstool-NG.sh@56]
    [ERROR]  >>
    [ERROR]  >>  For more info on this error, look at the file: 'build.log'
    [ERROR]  >>  There is a list of known issues, some with workarounds, in:
    [ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
    [ERROR]   
    [ERROR]  (elapsed: 24405467:06.85)
    [00:00] / ct-ng:152: recipe for target 'build' failed
    make[2]: *** [build] Error 1
    make[2]: Leaving directory '/home/frederik/esp-open-sdk/crosstool-NG'
    ../Makefile:116: recipe for target '_toolchain' failed
    make[1]: *** [_toolchain] Error 2
    make[1]: Leaving directory '/home/frederik/esp-open-sdk/crosstool-NG'
    Makefile:112: recipe for target '/home/frederik/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
    make: *** [/home/frederik/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

Still fails...

Looking at the file where that error message appears, it seems there are a lot of variables I shouldn't set... Can you update the README to explain when and how (and why) I should not set these variables?

ethe, 2016-05-31:
Also got this error

cranphin, 2016-10-20:
The error comes from crosstool-NG, not esp-open-sdk itself.
A very quick search gives me this file, which is where they originate, and which lists all the variables you should not set:
https://github.com/jcmvbkbc/crosstool-NG/blob/master/scripts/crosstool-NG.sh.in