pfalcon/esp-open-sdk #175: Build failed in step 'Installing cross-gdb'

fvpalha, 2016-06-08:
Hi.

I follow the instructions, but this error occurs:

```
[INFO ]  Installing cross-gdb
[ERROR]    configure: error: python is missing or unusable
[ERROR]    make[4]: *** [configure-gdb] Error 1
[ERROR]    make[3]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 26:36.59)
[26:37] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/palha/esp8266/esp-open-sdk/crosstool-NG'
../Makefile:116: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/palha/esp8266/esp-open-sdk/crosstool-NG'
Makefile:112: recipe for target '/home/palha/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/palha/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```

How I can fix this error?

likewise, 2016-06-08:
Hello,

there seems to be a development host dependency missing. I encountered the same error, I have the following fix:

Not mentioned but required to build gdb-7.1.0:

sudo apt-get install python-dev

Regards,

Leon.

fvpalha, 2016-06-08:
Thank you @likewise 
The tip works very well.

closed 2016-06-09 by fvpalha

kpishere, 2019-11-09:
Ah!  I had this issue.  It was because I had '/Library/Frameworks/Python.framework/Versions/2.7/bin'  in my path.  I forget which, but it looks like it was a common installer's behaviour to update MACOS/OSX's python.  Further to that, I think this path got parsed incorrectly in the config scripts.  Removing that from my path fixed the issue.  I was building on 10.13.6 HighSierra.

DasVoelkel, 2019-12-19:
This problem as occured for me too, i am on Archlinux and I tried everything I could find but it seems noone really knows what is wrong here...